vortexgpgpu/vortex #3: make error

zhangliang1997, 2020-11-30:
hi vortex team.
  i have installed verilator, riscv-tool-chain, but when i make vecadd, there is a error:
----------------------error------------------------------
asc@ubuntu:~/vortex/vortex/benchmarks/opencl/vecadd$ make
g++ -std=c++11 -O2 -Wall -Wextra -Wfatal-errors -Wno-deprecated-declarations -Wno-unused-parameter -I/opt/pocl/runtime/include -MM main.cc > .depend;
g++ -std=c++11 -O2 -Wall -Wextra -Wfatal-errors -Wno-deprecated-declarations -Wno-unused-parameter -I/opt/pocl/runtime/include main.cc -L/opt/pocl/runtime/lib -L/home/asc/vortex/vortex/driver/stub -lOpenCL -lvortex -o vecadd
/usr/bin/ld: cannot find -lOpenCL
/usr/bin/ld: cannot find -lvortex
---------------------------------------------------------
i know i need to install opencl, but what's the vortex and pocl/runtime/xxxx ?
can you give me some advice? thanks :)

felsabbagh3, 2020-12-01:
Hello @zhangliang1997. 

To be able to run OpenCL programs, the two things required are the OpenCL library (provided through POCL) and the vortex runtime kernel.

Vortex runtime kernel could be found under vortex/runtime. There's a makefile that will generate a dynamic library that is linked through -lvortex.

Vortex POCL could be found in this repo: https://github.com/vortexgpgpu/pocl
Please take a look at the readme.vortex in pocl repo to be able to compile it. That will also create a dynamic library that you can link through -lOpenCL

Please also keep in mind that you can utilize https://github.com/vortexgpgpu/vortex-toolchain-prebuilt repo instead of building everything from scratch.

Hope this helps!

zhangliang1997, 2020-12-03:
> Hello @zhangliang1997.
>
> To be able to run OpenCL programs, the two things required are the OpenCL library (provided through POCL) and the vortex runtime kernel.
>
> Vortex runtime kernel could be found [...]

thanks very much, i will try it.

closed 2020-12-03 by zhangliang1997