vortexgpgpu/vortex #5: how to enable branch divergence?

zhangliang1997, 2020-12-08:
Hello vortex team:
I've runed benchmarks/opencl/vecadd successfully. Now i modified the kernel.cl in vecadd,but it compiled failed.

The modified code(opencl/vecadd/kernel.cl):
```
#define __if(cond) split(cond); \
if(cond)
#define __endif join();

__kernel void vecadd (__global const float *A,
	                    __global const float *B,
	                    __global float *C)
{
  int gid = get_global_id(0);
  __if(id<2){
    C[gid] = A[gid] + B[gid];
  }else{
    C[gid] = A[gid] - B[gid];
  }__endif
}
```
Error info:

LLVM_PREFIX=/opt/llvm-riscv POCL_DEBUG=all LD_LIBRARY_PATH=/opt/llvm-riscv/lib:/opt/pocl/compiler/lib /opt/pocl/compiler/bin/poclcc -LLCFLAGS "-O3 -march=riscv32 -target-abi=ilp32f -mcpu=generic-rv32 -mattr=+m,+f -float-abi=hard -code-model=small" -CFLAGS "-v -O3 --sysroot=/opt/riscv-gnu-toolchain/riscv32-unknown-elf --gcc-toolchain=/opt/riscv-gnu-toolchain -march=rv32imf -mabi=ilp32f -I/home/zl/Desktop/riscv/vortex/runtime/include -fno-rtti -fno-exceptions -ffreestanding -nostartfiles -Wl,--gc-sections -Wl,-Bstatic,-T/home/zl/Desktop/riscv/vortex/runtime/linker/vx_link.ld" -LDFLAGS "/home/zl/Desktop/riscv/vortex/runtime/libvortexrt.a -lm" -o kernel.pocl kernel.cl
** Final POCL_DEBUG flags: FFFFFFFFFFFFFFFF 
[2020-11-08 05:48:10.067159997]POCL: in fn cl_int pocl_vortex_init(unsigned int, cl_device_id, const char*) at line 232:
  |   WARNING |  INIT dlcache DOTO delete
[2020-11-08 05:48:10.067414147]POCL: in fn compile_and_link_program at line 571:
  |      LLVM |  building program with options (null)
[2020-11-08 05:48:10.067420103]POCL: in fn compile_and_link_program at line 635:
  |   GENERAL |  building from sources for device 0
[2020-11-08 05:48:10.070914292]POCL: in fn int pocl_llvm_build_program(cl_program, unsigned int, const char*, char*, cl_uint, _cl_program* const*, const char**, int) at line 372:
  |      LLVM |  all build options: -Dcl_khr_global_int32_base_atomics -Dcl_khr_local_int32_base_atomics -Dcl_khr_3d_image_writes -Dcl_khr_fp16 -Dcl_khr_fp64 -cl-ext=-all,+cl_khr_global_int32_base_atomics,+cl_khr_local_int32_base_atomics,+cl_khr_3d_image_writes,+cl_khr_fp16,+cl_khr_fp64 -Dcl_khr_int64 -DPOCL_DEVICE_ADDRESS_BITS=32 -D__USE_CLANG_OPENCL_C_H -xcl -Dinline= -I. -cl-kernel-arg-info  -D__ENDIAN_LITTLE__=1 -D__IMAGE_SUPPORT__=1 -DCL_DEVICE_MAX_GLOBAL_VARIABLE_SIZE=0 -D__OPENCL_VERSION__=120 -cl-std=CL1.2 -D__OPENCL_C_VERSION__=120 -fno-builtin -triple=riscv32 -target-feature +m -target-feature +f -target-abi ilp32f 
### options: -Dcl_khr_global_int32_base_atomics -Dcl_khr_local_int32_base_atomics -Dcl_khr_3d_image_writes -Dcl_khr_fp16 -Dcl_khr_fp64 -cl-ext=-all,+cl_khr_global_int32_base_atomics,+cl_khr_local_int32_base_atomics,+cl_khr_3d_image_writes,+cl_khr_fp16,+cl_khr_fp64 -Dcl_khr_int64 -DPOCL_DEVICE_ADDRESS_BITS=32 -D__USE_CLANG_OPENCL_C_H -xcl -Dinline= -I. -cl-kernel-arg-info  -D__ENDIAN_LITTLE__=1 -D__IMAGE_SUPPORT__=1 -DCL_DEVICE_MAX_GLOBAL_VARIABLE_SIZE=0 -D__OPENCL_VERSION__=120 -cl-std=CL1.2 -D__OPENCL_C_VERSION__=120 -fno-builtin -triple=riscv32 -target-feature +m -target-feature +f -target-abi ilp32f user_options: 
### TargetOptions: Triple: riscv32, CPU: , ABI: ilp32f, Features: 
4 errors generated.
[2020-11-08 05:48:13.317627456]POCL: in fn void appendToProgramBuildLog(cl_program, unsigned int, std::__cxx11::string&) at line 141:
  |     ERROR |  error: /home/zl/.cache/pocl/kcache/tempfile-ee-9c-fc-a7-0e.cl:10:3 <Spelling=/home/zl/.cache/pocl/kcache/tempfile-ee-9c-fc-a7-0e.cl:1:20>: implicit declaration of function 'split' is invalid in OpenCL
error: /home/zl/.cache/pocl/kcache/tempfile-ee-9c-fc-a7-0e.cl:10:3 <Spelling=/home/zl/.cache/pocl/kcache/tempfile-ee-9c-fc-a7-0e.cl:10:8>: use of undeclared identifier 'id'
error: /home/zl/.cache/pocl/kcache/tempfile-ee-9c-fc-a7-0e.cl:10:3 <Spelling=/home/zl/.cache/pocl/kcache/tempfile-ee-9c-fc-a7-0e.cl:10:8>: use of undeclared identifier 'id'
error: /home/zl/.cache/pocl/kcache/tempfile-ee-9c-fc-a7-0e.cl:14:4 <Spelling=/home/zl/.cache/pocl/kcache/tempfile-ee-9c-fc-a7-0e.cl:3:17>: implicit declaration of function 'join' is invalid in OpenCL
[2020-11-08 05:48:13.322485175]POCL: in fn compile_and_link_program at line 641:
  |     ERROR | build_error_code pocl_llvm_build_program() failed
CL_BUILD_PROGRAM_FAILURE in main on line 371
Makefile:30: recipe for target 'kernel.pocl' failed
make: *** [kernel.pocl] Error 1

Something i did wrong? Can give me some advice?

zhangliang1997, 2020-12-10:
Hi vortex team, i read paper about [vortex](https://carrv.github.io/2019/papers/carrv2019_paper_10.pdf) and i found code some code
```
#define __if(cond) \
  register bool b asm("t0") = cond; \
  asm("split t0"); \
if (b) {
#define __else \
} else {
#define __endif \
} \
asm("join");
```
But it also can't be compiled successfully.

felsabbagh3, 2020-12-10:
Hey @zhangliang1997, the code for this paper was just to illustrate a point, how it's actually implemented in software is using the two functions called vx_split and vx_join. These two functions are defined in runtime/src/vx_intrinsics.S

Also, please note that __if and __endif are already defined in runtime/include/vx_intrinsics.h:
```
#define __if(b) vx_split(b); \
                if (b) 

#define __else else

#define __endif vx_join();
```

Also please note that on [http://vortex.cc.gatech.edu/](http://vortex.cc.gatech.edu/) you will find more recent papers on Vortex.

zhangliang1997, 2020-12-10:
> Hey @zhangliang1997, the code for this paper was just to illustrate a point, how it's actually implemented in software is using the two functions called vx_split and vx_join. These two functions are [...]

closed 2020-12-10 by zhangliang1997

zhangliang1997, 2020-12-10:
hi @felsabbagh3, i have modified vecadd/kernel.cl
```
#include "../../../runtime/include/vx_intrinsics.h"
__kernel void vecadd (__global const float *A,
	                    __global const float *B,
	                    __global float *C)
{
  int gid = get_global_id(0);
  __if(gid<2){
    C[gid] = A[gid] + B[gid];
  }__else{
    C[gid] = A[gid] - B[gid];
  }__endif
}
```
But still have a error. I think i  need  read more code of vortex.
r=+m,+f -float-abi=hard -code-model=small" -CFLAGS "-v -O3 --sysroot=/opt/riscv-gnu-toolchain/riscv32-unknown-elf --gcc-toolchain=/opt/riscv-gnu-toolchain -march=rv32imf -mabi=ilp32f -I/home/zl/Desktop/riscv/vortex/runtime/include -fno-rtti -fno-exceptions -ffreestanding -nostartfiles -Wl,--gc-sections -Wl,-Bstatic,-T/home/zl/Desktop/riscv/vortex/runtime/linker/vx_link.ld" -LDFLAGS "/home/zl/Desktop/riscv/vortex/runtime/libvortexrt.a -lm" -o kernel.pocl kernel.cl
** Final POCL_DEBUG flags: FFFFFFFFFFFFFFFF 
[2020-11-10 06:24:42.514002416]POCL: in fn cl_int pocl_vortex_init(unsigned int, cl_device_id, const char*) at line 232:
  |   WARNING |  INIT dlcache DOTO delete
[2020-11-10 06:24:42.515297264]POCL: in fn compile_and_link_program at line 571:
  |      LLVM |  building program with options (null)
[2020-11-10 06:24:42.515334827]POCL: in fn compile_and_link_program at line 635:
  |   GENERAL |  building from sources for device 0
[2020-11-10 06:24:42.532420117]POCL: in fn int pocl_llvm_build_program(cl_program, unsigned int, const char*, char*, cl_uint, _cl_program* const*, const char**, int) at line 372:
  |      LLVM |  all build options: -Dcl_khr_global_int32_base_atomics -Dcl_khr_local_int32_base_atomics -Dcl_khr_3d_image_writes -Dcl_khr_fp16 -Dcl_khr_fp64 -cl-ext=-all,+cl_khr_global_int32_base_atomics,+cl_khr_local_int32_base_atomics,+cl_khr_3d_image_writes,+cl_khr_fp16,+cl_khr_fp64 -Dcl_khr_int64 -DPOCL_DEVICE_ADDRESS_BITS=32 -D__USE_CLANG_OPENCL_C_H -xcl -Dinline= -I. -cl-kernel-arg-info  -D__ENDIAN_LITTLE__=1 -D__IMAGE_SUPPORT__=1 -DCL_DEVICE_MAX_GLOBAL_VARIABLE_SIZE=0 -D__OPENCL_VERSION__=120 -cl-std=CL1.2 -D__OPENCL_C_VERSION__=120 -fno-builtin -triple=riscv32 -target-feature +m -target-feature +f -target-abi ilp32f 
### options: -Dcl_khr_global_int32_base_atomics -Dcl_khr_local_int32_base_atomics -Dcl_khr_3d_image_writes -Dcl_khr_fp16 -Dcl_khr_fp64 -cl-ext=-all,+cl_khr_global_int32_base_atomics,+cl_khr_local_int32_base_atomics,+cl_khr_3d_image_writes,+cl_khr_fp16,+cl_khr_fp64 -Dcl_khr_int64 -DPOCL_DEVICE_ADDRESS_BITS=32 -D__USE_CLANG_OPENCL_C_H -xcl -Dinline= -I. -cl-kernel-arg-info  -D__ENDIAN_LITTLE__=1 -D__IMAGE_SUPPORT__=1 -DCL_DEVICE_MAX_GLOBAL_VARIABLE_SIZE=0 -D__OPENCL_VERSION__=120 -cl-std=CL1.2 -D__OPENCL_C_VERSION__=120 -fno-builtin -triple=riscv32 -target-feature +m -target-feature +f -target-abi ilp32f user_options: 
### TargetOptions: Triple: riscv32, CPU: , ABI: ilp32f, Features: 
[2020-11-10 06:24:45.934403144]POCL: in fn llvm::Module* getKernelLibrary(cl_device_id) at line 892:
  |      LLVM |  Using /opt/pocl/compiler/lib/../share/pocl/kernel-riscv32.bc as the built-in lib.
linker:search: vecadd calls _Z13get_global_idj
linker:search: recursing into _Z13get_global_idj
linker:it's a declaration.
linker:search: vecadd calls vx_split
linker:search: recursing into vx_split
linker:it's a declaration.
linker:search: vecadd calls vx_join
linker:search: recursing into vx_join
linker:it's a declaration.
linker:_Z13get_global_idj is not defined
linker:vx_split is not defined
linker:vx_join is not defined
linker:cloning the global variables:
linker: _global_offset_x
linker: _local_size_x
linker: _group_id_x
linker: _global_offset_y
linker: _local_size_y
linker: _group_id_y
linker: _global_offset_z
linker: _local_size_z
linker: _group_id_z
linker: _num_groups_x
linker: _num_groups_y
linker: _num_groups_z
linker: _local_id_x
linker: _local_id_y
linker: _local_id_z
linker: _work_dim
linker: __const.__pocl_print_floats_float.NANs
linker: __const.__pocl_print_floats_float.INFs
linker: .str.2
linker: .str.3
linker: .str
linker: .str.1
linker: .str.4
linker: .str.5
linker: _printf_buffer
linker: _printf_buffer_capacity
linker: _printf_buffer_position
linker:copying function _Z13get_global_idj with callgraph
linker:search: _Z13get_global_idj calls _Z12get_local_idj
linker:search: recursing into _Z12get_local_idj
linker:search: _Z13get_global_idj calls _Z12get_local_idj
linker:search: _Z13get_global_idj calls _Z12get_local_idj
linker:copying function _Z12get_local_idj with callgraph
linker:   _Z12get_local_idj not found in destination module, creating
linker:  cloning _Z12get_local_idj
linker:  cloning _Z13get_global_idj
[2020-11-10 06:24:47.288086114]POCL: in fn void appendToProgramBuildLog(cl_program, unsigned int, std::__cxx11::string&) at line 141:
  |     ERROR |  Error(s) while linking: 
###Cannot find symbol vx_join in kernel library
###Cannot find symbol vx_split in kernel library
[2020-11-10 06:24:47.293096956]POCL: in fn compile_and_link_program at line 641:
  |     ERROR | build_error_code pocl_llvm_build_program() failed
CL_BUILD_PROGRAM_FAILURE in main on line 371
Makefile:30: recipe for target 'kernel.pocl' failed
make: *** [kernel.pocl] Error 1

By the way, i'm very interested in vortex and i've sended a email to vortexgpu@cc.gatech.edu. If there's a chance, i want to do something for vortex. 
Thanks for your replay:)