vortexgpgpu/vortex #6: Is it possible to contribute outside Georgia Tech?

avl-bsuir, 2021-01-17:
Hi!

I think this is amazing project to work with. I am happy to see that it is also under active development. Unfortunately, current state of documentation does not allow to understand where the project is moving. Is it possible to contribute to the project for someone who is not a part of your research group at Georgia Tech? I understand, that you may study some novel features you'd like to publish eventually, but maybe you have non-research tasks?

Thank you,
Anton.

tinebp, 2021-01-17:
We are looking for people who can help with the development effort.Are you a student?
regards,-Blaise

avl-bsuir, 2021-01-18:
I am assistant prof. in BSUIR. I teach Computer Architecture for undergrads and a couple of them might join you for a semester under my supervision.

tinebp, 2021-01-25:
We would love to have undergraduate/graduate students work on mini-projects with us.
We have had undergraduate/graduate students every year working on Vortex.
Could you please contact Prof. Kim how is our director of the project?
hyesoon@cc.gatech.edu
regards,-Blaise

closed 2021-04-06 by hyesoon