vortexgpgpu/vortex #9: Error running  "Quick Test running OpenCL vecadd program"

hossamfadeel, 2021-04-02:
Hello,

I followed the steps o building vortex and when I tried to test vecadd program I got the following:

ubuntu@ubuntu1804:~/vortex/benchmarks/opencl/vecadd$ make
make: Nothing to be done for 'all'.
ubuntu@ubuntu1804:~/vortex/benchmarks/opencl/vecadd$ sudo make
make: Nothing to be done for 'all'.
ubuntu@ubuntu1804:~/vortex/benchmarks/opencl/vecadd$ make run-rtlsim
LD_LIBRARY_PATH=/opt/pocl/runtime/lib:/home/ubuntu/vortex/driver/rtlsim: ./vecadd -n64
OpenCL Error: 'clGetPlatformIDs(1, &platform_id, NULL)' returned -1001!
Makefile:50: recipe for target 'run-rtlsim' failed
make: *** [run-rtlsim] Error 255

Could you help me figure out what is going on.??

Thanks in advace

hossamfadeel, 2021-04-05:
**Solved by increasing memory size of my VM.**

ubuntu@ubuntu1804:~/vortex/benchmarks/opencl/vecadd$ make run-rtlsim
LD_LIBRARY_PATH=/opt/pocl/runtime/lib:/home/ubuntu/vortex/driver/rtlsim: ./vecadd -n64
Create context
Allocate device buffers
Create program from kernel source
Upload source buffers
Execute the kernel
Elapsed time: 42 ms
Download destination buffer
Verify result
PASSED!
PERF: instrs=4897, cycles=6048, IPC=0.809689

Thanks

hossamfadeel, 2021-04-05:
Passed Examples:
vecadd
sgemm

## Need Elaboration:
**BlackScholes**
ubuntu@ubuntu1804:~/vortex/benchmarks/opencl/BlackScholes$ make
make: *** No rule to make target 'main.cc', needed by 'BlackScholes.elf'.  Stop.
ubuntu@ubuntu1804:~/vortex/benchmarks/opencl/BlackScholes$ make run-rtlsim
make: *** No rule to make target 'run-rtlsim'.  Stop.

**DotProduct**
ubuntu@ubuntu1804:~/vortex/benchmarks/opencl/DotProduct$ make
POCL_DEBUG=all POCL_DEBUG_LLVM_PASSES=1 LD_LIBRARY_PATH=/lib:../../../../pocl/drops_riscv_cc/lib ../../../../pocl/drops_riscv_cc/bin/poclcc -o libDotProduct.a kernel.cl
IO error on file kernel.cl: No such file or directory
Makefile:38: recipe for target 'libDotProduct.a' failed
make: *** [libDotProduct.a] Error 2
ubuntu@ubuntu1804:~/vortex/benchmarks/opencl/DotProduct$

wickedb, 2021-10-20:
Thanks!! I'll let you know if this works.

wickedb, 2021-11-01:
Thank you!! This worked for me after I installed the toolchain AND set the right env variables (Verilator, pocl, riscvtoolchain, etc.). Thanks a lot again.