vortexgpgpu/vortex #10: Local memory issue

ToT-CaMbIu, 2021-04-12:
Hi!
I’m currently working on a neural network framework for execution on vortexgpu. I’ve faced a problem while implementing kernels with local memory usage. When I use static allocated memory, the program executes incorrectly for some reasons. For example, for this kernel
```
__kernel void sample(int n, int m,
                     const __global float *A,
                     __global float *C) {
    const int rowLocal = get_local_id(0);
    const int colLocal = get_local_id(1);
    const int rowGlobal = get_global_id(0);
    const int colGlobal = get_global_id(1);
    
    __local float buf[4];
    
    buf[rowLocal * 2 + colLocal] = A[rowGlobal * m + colGlobal];

    barrier(CLK_LOCAL_MEM_FENCE);

    C[rowGlobal * m + colGlobal] = buf[rowLocal * 2 + colLocal];
}
```

i've got this result:

```
A:
0 1 2 3
4 5 6 7
8 9 10 11
12 13 14 15
Device ready...
Create context
Create program from kernel source
Upload source buffers
Execute the kernel
Device running...
Device ready...
C:
10 11 10 11
14 15 14 15
10 11 10 11
14 15 14 15
kernels took 1.009008 seconds to execute
```
If I use dynamic allocated buffer, i’m facing an abortion after clGetDeviceIDs execution (cannot even trace an error for some reason).

Can anyone suggest workarounds for this behaviour? Do you support local memory? Maybe there is a version that supports it?
 
Thank you,
Kirill.

blobbiesAI, 2022-09-14:
Did you solve this problem, I had a similar problem and was wondering if local memory is supported?

lpc97667, 2024-06-26:
Hi,

Local memory is now supported.