vortexgpgpu/vortex #17: main.cc:5:10: fatal error: CL/opencl.h: No such file or directory

nicolast0604, 2021-07-19:
make[3]: Entering directory '/home/nicolast0604/vortex/tests/opencl/vecadd'
g++ -std=c++11 -O2 -Wall -Wextra -Wfatal-errors -Wno-deprecated-declarations -Wno-unused-parameter -I/opt/pocl/runtime/include -MM main.cc > .depend;
g++ -std=c++11 -O2 -Wall -Wextra -Wfatal-errors -Wno-deprecated-declarations -Wno-unused-parameter -I/opt/pocl/runtime/include main.cc -L/opt/pocl/runtime/lib -L/home/nicolast0604/vortex/driver/stub -lOpenCL -lvortex -o vecadd
main.cc:5:10: fatal error: CL/opencl.h: No such file or directory
 #include <CL/opencl.h>
          ^~~~~~~~~~~~~
compilation terminated.
Makefile:35: recipe for target 'vecadd' failed
make[3]: *** [vecadd] Error 1
make[3]: Leaving directory '/home/nicolast0604/vortex/tests/opencl/vecadd'
Makefile:2: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/home/nicolast0604/vortex/tests/opencl'
Makefile:10: recipe for target 'opencl' failed
make[1]: *** [opencl] Error 2
make[1]: Leaving directory '/home/nicolast0604/vortex/tests'
Makefile:3: recipe for target 'all' failed
make: *** [all] Error 2

Any hint?

hossamfadeel, 2021-08-25:
Did you install POCL??

Also, you can try running vecadd using the following command: (**_Note_**: **driver** can be **vlsim** or **rtlsim** or **simx**)
./ci/blackbox.sh --clusters=1 --cores=4 --warps=4 --threads=4 --driver=vlsim --app=vecadd

lpc97667, 2024-06-26:
Typically these types of OpenCL errors are due to a problem with the PoCL install or environment. Closing due to staleness.

closed 2024-06-26 by lpc97667