vortexgpgpu/vortex #19: Error while building Quartus project for Vortex

hossamfadeel, 2021-09-08:
Hello,

I am trying to build a Quartus project using the provided make file for Quartus tcl file

`vortex/hw/syn/quartus/$**make vortex**`

I tried others and I am getting the same error ... Do you have any idea why it complain about **make_all_pins_virtual** in the project.tcl?
```
Error (23035): Tcl error: ERROR: Error(s) found while running an executable. See report file(s) for error message(s). Message log indicates which executable was run last.

    while executing
"execute_module -tool map"
    (procedure "make_all_pins_virtual" line 2)
    invoked from within
"make_all_pins_virtual"
    (file "../../project.tcl" line 101)
------------------------------------------------
ERROR: Error(s) found while running an executable. See report file(s) for error message(s). Message log indicates which executable was run last.

    while executing
"execute_module -tool map"
    (procedure "make_all_pins_virtual" line 2)
    invoked from within
"make_all_pins_virtual"
    (file "../../project.tcl" line 101)
------------------------------------------------
Error (23031): Evaluation of Tcl script ../../project.tcl unsuccessful
Error: Quartus Prime Shell was unsuccessful. 2 errors, 20 warnings
    Error: Peak virtual memory: 1136 megabytes
    Error: Processing ended: Wed Sep  8 15:21:17 2021
    Error: Elapsed time: 00:00:04
Makefile:61: recipe for target 'vortex_afu.qpf' failed
make[1]: *** [vortex_afu.qpf] Error 3
make[1]: Leaving directory '/home/sa/vortexgpgpu/vortex/hw/syn/quartus/top2/build'
```

Thanks in advance

wickedb, 2022-01-04:
I have problems like this too. 

Are you using intel devcloud? How did you set the `export OPAE_HOME=/opt/opae/1.1.2`? Did you install OPAE?

lpc97667, 2024-06-26:
The Quartus makefile in the current master should build the default 1-core configuration when `make` is ran in hw/syn/altera/opae. Closing due to staleness; please open a new issue if it persists.

closed 2024-06-26 by lpc97667