vortexgpgpu/vortex #20: can't link soft-float modules with single-float modules

federicorossifr, 2021-09-21:
I followed the installation guide, but when I try to compile using `make all` i get the following errors:

> ******/lib/gcc/riscv32-unknown-elf/11.1.0/../../../../riscv32-unknown-elf/bin/ld: *********/lib/gcc/riscv32-unknown-elf/11.1.0/libgcc.a(_clz.o): can't link soft-float modules with single-float modules
>******/lib/gcc/riscv32-unknown-elf/11.1.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file *****/lib/gcc/riscv32-unknown-elf/11.1.0/libgcc.a(_clz.o)

SoaringAnecdotalBear, 2021-10-17:
did you solved this error?

lpc97667, 2024-06-26:
The toolchain has since changed. Closing due to staleness; please open a new issue if it persists.

closed 2024-06-26 by lpc97667