vortexgpgpu/vortex #32: RTL synthesis error of hw/rtl using Xilinx Vivado

artyom-mankevich, 2021-12-01:
Failing to synthesize project containing hw/rtl and hw/dpi directories using Xilinx Vivado 2021.1. The following errors show up in [hw/rtl/fp_cores/VX_fpu_dpi.sv](https://github.com/vortexgpgpu/vortex/blob/d64bc880b23094489098bc51f86575e6c674f728/hw/rtl/fp_cores/VX_fpu_dpi.sv):
[image]
For some reason all float_dpi functions are considered undeclared in always procedural block at lines 131-141. I suspect this is happening since in the [hw/rtl/fp_cores/VX_fpu_define.vh](https://github.com/vortexgpgpu/vortex/blob/d64bc880b23094489098bc51f86575e6c674f728/hw/rtl/fp_cores/VX_fpu_define.vh) file the following if statement is defined which prevents float_dpi functions from being loaded during synthesis:
[image]
If I force include the float_dpi.vh in header file these errors appear:
[image]

I tried using Xilinx Vivado 2019.1 encountering the same issue. Device used for project is Xilinx Artix-7 FPGA AC701 Evaluation Kit, although I doubt this is of any significance as this problem persists with other boards too.
OS is Windows 10 build 19042.

If this is the intended behavior during synthesis, is there any other way to synthesize the project?
Thanks in advance!

Rizkan19, 2021-12-15:
Hi. 
I have the same problem, did you solve the issue or  did you find any other synthesis technique?

artyom-mankevich, 2021-12-15:
@Rizkan19
Hello! I noticed that since errors are in function calls it doesn't affect synthesised scheme for my purposes, so I just commented out all functions that have names starting with "dpi_" in VX_fpu_dpi.sv and then the synthesis worked correctly.

Rizkan19, 2021-12-15:
@artyom-mankevich 
Okey I will try. Thank you so much :)

closed 2024-06-26 by lpc97667

JinB1, 2024-11-19:
Hello, have you tested it on the FPGA after synthesis?