vortexgpgpu/vortex #39: Are any code modifications required to run the BFS test code?

wickedb, 2022-01-15:
I have the following error while running BFS, something similar is mentioned in #9

```bash
[568169] %Error: VX_csr_data.sv:229: Assertion failed in TOP.vortex_afu_shim.afu.vortex.genblk2[0].cluster.genblk2[0].core.pipeline.execute.csr_unit.csr_data: invalid CSR read address: 26
%Error: ../../hw/rtl/VX_csr_data.sv:229: Verilog $stop
Aborting...
Aborted (core dumped)
Makefile:40: recipe for target 'run-vlsim' failed
make: *** [run-vlsim] Error 134
```

Thanks in advance

closed 2022-03-02 by wickedb

nimish15shah, 2022-03-17:
I am also facing this issue. I have set the following env variables:

VERILATOR_ROOT
LLVM_PREFIX
POCL_CC_PATH
POCL_RT_PATH
RISCV_TOOLCHAIN_PATH
SYSROOT

But, I am still getting this error. 
I did not modify the DEVICE or PLATFORM ID & TYPE. Is that the issue?

nimish15shah, 2022-03-21:
Ok, I managed to solve the "invalid CSR read address: 26" assertion in the BFS code. Indeed, the problem was with the incorrect environment variable. The POCL_CC_PATH and POCL_RT_PATH were pointing to the ubuntu prebuilt toolchain, while I am on a centos 7 machine.

wickedb, 2022-06-20:
Did your code work finally? I couldn't solve the error so I ported the original Rodinia BFS code to Vortex.

Here's the code: https://github.com/wickedb/vortex/tree/master/tests/opencl/bfs2