vortexgpgpu/vortex #43: Wrong install location for verilator using install_toolchain.sh

saurabhsingh99100, 2022-02-09:
When running `./ci/toolchain_install.sh -all` on a fresh system to install prerequisites for the vortex project, it installs verilator at `/opt/verilator/`. However, when running `make` in the vortex project directory, it expects the Verilator path to be `/opt/verilator/share/verilator/`. This path mismatch gives errors in building vortex sources.

chenkaibytedance, 2022-03-02:
hi Saursin,
have you solve the issue

drwillwang, 2022-03-12:
before make, execute the following two setup:
export VERILATOR_ROOT=/opt/verilator
export PATH="/opt/verilator/bin:$PATH"

nimish15shah, 2022-03-16:
Just setting up the VERILATOR_ROOT and PATH did not solve the problem, as it still produces the following error:

`VVortex.mk:63: /opt/verilator/include/verilated.mk: No such file or directory`

I had to copy-paste the content of `/opt/verilator/share/verilator/` to `/opt/verilator/`

Note that I only faced this error while working with the centos7 prebuilt binaries. This issue does not occur with the ubuntu binaries.

yangzexia, 2022-05-11:
Have you resolved this problem?

yangzexia, 2022-05-11:
Have you solved this problem?

drwillwang, 2022-05-11:
Hi, Zexia,

 

Yes, I’ve solved the issue a while ago.

 

I’m wondering if you have some time for a short discussion on Vortex and toolchain.

 

Best regards,

Will Wang

yangzexia, 2022-05-11:
> 

Thank you, I solved the issue by following the hint of [[nimish15shah](https://github.com/nimish15shah)](https://github.com/vortexgpgpu/vortex/issues/43#issuecomment-1069234690) above. 

By the way, I have another problem and I'd appreciate if you can help me. 

I want to export the assembly text of kernel part of .cl file input, but I wonder if the pocl modifided have this function, do you have any idea? 

I tried to modify the "-CFLAGS" arguments of makefile, which directory is https://github.com/vortexgpgpu/vortex/blob/77002dd06ad51dd71497d2a17842c3f5b1989db6/tests/opencl/vecadd/Makefile#L40. 
I guess that -CFLAGS arguments are used by clang complier of OpenCL input, however I don't know how these arguments transform to clang, so I am viewing the pocl source code. Looking forward to your reply.

closed 2024-06-26 by lpc97667