vortexgpgpu/vortex #51: How to dump assembly text ?

yangzexia, 2022-05-12:
I want to dump the assembly .s file of device part of OpenCL source. I tried to add arguments like "-S" that often used in Clang to "-CFLAGS" of makefile below but it didn't work. 

https://github.com/vortexgpgpu/vortex/blob/77002dd06ad51dd71497d2a17842c3f5b1989db6/tests/opencl/vecadd/Makefile#L14-L40

vinitpawaskar, 2023-12-18:
Hi! Were you able to generate the .s assembly file of source or kernel. I'm also struggling to get it.

tinebp, 2026-06-14:
To disassemble the device code, run `llvm-objdump` from the Vortex LLVM toolchain on the compiled kernel ELF, e.g. `$LLVM_VORTEX/bin/llvm-objdump -d kernel.elf`. Most test Makefiles already emit a `kernel.dump` artifact during the build. Closing as answered.

closed 2026-06-14 by tinebp