vortexgpgpu/vortex #61: Verilator commmand exit with 2

deepelixir, 2022-09-17:
VVortex.mk:63: /opt/verilator/share/verilator/include/verilated.mk: No such file or directory
make[3]: *** No rule to make target '/opt/verilator/share/verilator/include/verilated.mk'.  Stop.
%Error: make -C obj_dir -f VVortex.mk -j 4 exitted with 2
%Error: Command Failed /opt/verilator/bin/verilator_bin --build --exe --cc Vortex --top-module Vortex -O2 --language 1800-2009 --assert -Wall -Wpedantic -Wno-DECLFILENAME -Wno-REDEFMACRO --x-initial unique --x-assign unique verilator.vlt -I../../hw/rtl -I../../hw/dpi -I../../hw/rtl/libs -I../../hw/rtl/interfaces -I../../hw/rtl/cache -I../../hw/rtl/simulate -I../../hw/rtl/fp_cores -I../../third_party/fpnew/src/common_cells/include -I../../third_party/fpnew/src/common_cells/src -I../../third_party/fpnew/src/fpu_div_sqrt_mvp/hdl -I../../third_party/fpnew/src -I../../hw/rtl/tex_unit -j 4 -DNDEBUG -DIMUL_DPI -DIDIV_DPI -DFPU_DPI ../common/util.cpp ../common/mem.cpp ../common/rvfloats.cpp ../../hw/dpi/util_dpi.cpp ../../hw/dpi/float_dpi.cpp processor.cpp main.cpp -CFLAGS -std\=c\+\+11\ -Wall\ -Wextra\ -Wfatal-errors\ -Wno-array-bounds\ -fPIC\ -Wno-maybe-uninitialized\ -I../../../hw\ -I../../common\ -I../../../third_party/softfloat/source/include\ -I../../../third_party\ \ -O2\ -DNDEBUG -LDFLAGS ../../../third_party/softfloat/build/Linux-x86_64-GCC/softfloat.a\ -L../../../third_party/ramulator\ -lramulator -o ../rtlsim
make[2]: *** [Makefile:88: rtlsim] Error 2
make[1]: *** [Makefile:3: all] Error 2
make: *** [Makefile:4: all] Error 2

BrianQian1999, 2022-10-10:
Hi, I'm facing the same problem. Have you solved it? 
I'm using <Verilator 4.219 devel rev v4.216-111-ge52a4ac7> and Ubuntu20.04.

omerguzelelectronicguy, 2022-11-16:
Hello,
I solved it by using a soft link. This was the solution for me after just getting this error.
```
cd /opt/verilator/
mkdir share;     cd share;
mkdir verilator; cd verilator;
ln -s ../../include/ include
```
After this fixation, there will be one more dependency problem(libtinfo) which I solved by the command below.
`sudo apt install libncurses5`

closed 2024-06-26 by lpc97667