vortexgpgpu/vortex #67: Possibly incorrect readme info

aeteijeiro, 2022-12-04:
In the read me file, it is stated that the supported FPGAs for this project are Arria 10/Stratix 10 FPGAs. However, from my research, it seems like not just any one will do - you need a card from the Intel PAC lineup with those FPGAs in order to use the intel acceleration stack that is implied in the FPGA setup guide. Can someone who has gotten this working please clear this up for all of us?

wickedb, 2023-07-02:
I got it working on Intel devcloud, it comes with all the necessary dependencies: [this doc](https://github.com/wickedb/vortex/blob/master/docs/intel_devcloud.md) might help

aeteijeiro, 2023-07-02:
Thank you so, so much. I must have tried every cloud provider but this one. I did not know Intel devcloud existed and it never showed up in my search results. Thank you so much for making a fork of this and I will try it out this week!

aeteijeiro, 2023-07-08:
@wickedb I got it working, using your guide. Thanks again! On to see what I can do with it!

closed 2023-07-08 by aeteijeiro

wickedb, 2023-09-12:
@aeteijeiro Happy to help! Would you mind changing your issue to include the word "FPGA"? It would make it easier for someone searching for the same solution.

aeteijeiro, 2023-09-17:
For sure. I just amended it.