vortexgpgpu/vortex #89: Couldn't run openCL vecadd demo on vortex2.0: /usr/bin/ld: couldn't find -lvortex: No such file or directory

yff18, 2023-10-24:
dear vortex team:
  I encountered this problem when running the test case, here's the log information:
`./ci/blackbox.sh --cores=2 --app=vecadd
CONFIGS=-DNUM_CLUSTERS=1 -DNUM_CORES=2 -DNUM_WARPS=4 -DNUM_THREADS=4    
running: CONFIGS=-DNUM_CLUSTERS=1 -DNUM_CORES=2 -DNUM_WARPS=4 -DNUM_THREADS=4  make -C ./ci/../runtime/simx
running: make -C ./ci/../tests/opencl/vecadd run-simx
make: entering directory“/home/yff/vortex/tests/opencl/vecadd”
g++ -std=c++11 -Wall -Wextra -Wfatal-errors -Wno-deprecated-declarations -Wno-unused-parameter -Wno-narrowing -I/opt/pocl/runtime/include -O2 -DNDEBUG -c main.cc -o main.cc.o
In file included from /opt/pocl/runtime/include/CL/cl.h:20,
                 from /opt/pocl/runtime/include/CL/opencl.h:24,
                 from main.cc:5:
/opt/pocl/runtime/include/CL/cl_version.h:22:104: note: ‘#pragma message: cl_version.h: CL_TARGET_OPENCL_VERSION is not defined. Defaulting to 300 (OpenCL 3.0)’
   22 | #pragma message("cl_version.h: CL_TARGET_OPENCL_VERSION is not defined. Defaulting to 300 (OpenCL 3.0)")
      |                                                                                                        ^
g++ -std=c++11 -Wall -Wextra -Wfatal-errors -Wno-deprecated-declarations -Wno-unused-parameter -Wno-narrowing -I/opt/pocl/runtime/include -O2 -DNDEBUG main.cc.o -L/opt/pocl/runtime/lib -L~/vortex/runtime/stub -lvortex /opt/pocl/runtime/lib/libOpenCL.so	 -o vecadd
/usr/bin/ld:couldn't find -lvortex: no such file or directory
collect2: error: ld returned 1 exit status
make: *** [../common.mk:87：vecadd] error 1
make: leaving directory“/home/yff/vortex/tests/opencl/vecadd”
`
  I follow the steps in README, did I miss some configuration?
  thanks

tinebp, 2023-10-24:
the migration to vortex 2.0 requires a clean setup and reinstalling all dependencies. 
follow instructions in README.

yff18, 2023-10-24:
> the migration to vortex 2.0 requires a clean setup and reinstalling all dependencies. follow instructions in README.

yes, I re-clone the project source code and re-install the toolchain, but still get the same result (no such file or directory).

tinebp, 2023-10-24:
/usr/bin/ld:couldn't find -lvortex: no such file or directory.

this indicates that libvortex is missing on your build.
Per README instructions you need to run the following command to completion before running any test

$ make -s

tinebp, 2024-08-02:
let us know if you are still facing the issue

closed 2024-08-02 by tinebp