vortexgpgpu/vortex #90: /opt/pocl/runtime/lib/libOpenCL.so: undefined reference to `hwloc_topology_ignore_type'

PromiseMax97, 2023-10-29:
$ make -s
/usr/bin/ld: /opt/pocl/runtime/lib/libOpenCL.so: undefined reference to `hwloc_topology_ignore_type'
collect2: error: ld returned 1 exit status
make[3]: *** [../common.mk:87: vecadd] Error 1
make[2]: *** [Makefile:2: all] Error 2
make[1]: *** [Makefile:11: opencl] Error 2
make: *** [Makefile:7: all] Error 2
I follow the steps in README, but I encountered the above problem when executing "make -s"，did I miss some configuration?
thanks

rodhuega, 2023-11-06:
Hi, I'm facing the same problem.
I have tried to install hwloc and libhwloc-dev. Also to create symbolic links to point to libhwloc.so.5. Even I downloaded and installed by hand hwloc with the following commands: `wget https://download.open-mpi.org/release/hwloc/v2.4/hwloc-2.4.0.tar.gz && tar -xzvf hwloc-2.4.0.tar.gz && cd hwloc-2.4.0 && ./configure && make && make install` but it didn't work too. My Ubuntu version is 20.04.6 LTS.

chbcr, 2023-11-07:
Using older versions of HW, such as V1.10.1

rodhuega, 2023-11-07:
Thanks, using the version that you indicated (downloaded from [here](https://www.open-mpi.org/software/hwloc/v1.10/)) has worked and now I can compile successfully.

HakamAtassi, 2023-11-08:
Thanks @rodhuega, 

Installing the version linked (and adding export LD_LIBRARY="home/<usr>/hwloc-1.10.1/src/.libs/:$LD_LIBRARY" to .bashrc) fixed the build and now runs the provided test.

chbcr, 2023-11-08:
HI, Another question, have you encountered this issue when compiling RTLSIM?
make -C rtlsim
%Error: ../../hw/rtl/interfaces/VX_warp_ctl_if.sv:16:26: syntax error, unexpected import, expecting ';'
   16 | interface VX_warp_ctl_if import VX_gpu_pkg::*; ();
      |                          ^~~~~~
        ../../hw/rtl/core/VX_core.sv:60:1: ... note: In file included from VX_core.sv
        ../../hw/rtl/VX_socket.sv:224:1: ... note: In file included from VX_socket.sv
        ../../hw/rtl/VX_cluster.sv:149:1: ... note: In file included from VX_cluster.sv
        ../../hw/rtl/Vortex.sv:150:1: ... note: In file included from Vortex.sv
%Error: ../../hw/rtl/interfaces/VX_warp_ctl_if.sv:16:48: syntax error, unexpected '('
   16 | interface VX_warp_ctl_if import VX_gpu_pkg::*; ();
      |                                                ^
        ../../hw/rtl/core/VX_core.sv:60:1: ... note: In file included from VX_core.sv
        ../../hw/rtl/VX_socket.sv:224:1: ... note: In file included from VX_socket.sv
        ../../hw/rtl/VX_cluster.sv:149:1: ... note: In file included from VX_cluster.sv
        ../../hw/rtl/Vortex.sv:150:1: ... note: In file included from Vortex.sv
%Error: ../../hw/rtl/interfaces/VX_warp_ctl_if.sv:26:5: syntax error, unexpected modport
   26 |     modport master (

rodhuega, 2023-11-08:
For me is working properly. I tested doing a `cd sim && make clean -C rtlsim && make -C rtlsim` and the compilation was completed successfully.

chbcr, 2023-11-14:
Okay, thank you all the same

closed 2023-11-20 by PromiseMax97