vortexgpgpu/vortex #97: does anyone know the implementation of cache coherence in vortex?

BeenLi, 2024-01-04:
the paper "Vortex: Extending the RISC-V ISA for GPGPU and 3D-Graphics" mentions that "Flush operations among caches are provided as a means of providing weak coherent memory space.". And the [cache_subsystem.md] (https://github.com/vortexgpgpu/vortex/blob/master/docs/cache_subsystem.md) in docs/cache_subsystem mentions that "Caches communicate via snooping" and "Cache flush from AFU is passed down the hierarchy​".

[image]

But I can't find the flush operations and snoop operations in the source code(release_v2) located in the `/hw/rtl/cache` directory.

Ifoufou, 2024-01-09:
I'm not part of the Vortex dev team so take the following with a grain of salt.

I think they removed the snoop protocol ensuring cache coherency a while ago (between the version tagged 0.1.0 and 0.2.0, at this commit https://github.com/vortexgpgpu/vortex/commit/703a861fe9dd120753c587e86428ebab3a3320ab). As you quoted, it was possible at that time to trigger invalidations with the device API (AFU).

Today, my understanding is that there is no coherence between the private L1 data-caches, between the different instances of the L2, and also hierarchically (L1 <--> L2 <--> L3) i.e. you cannot safely exchange values between threads located in different Vortex cores, nor in different Vortex sockets. Indeed, every cache is write-through but it is not sufficient to prevent the read of old stale data (unless they made a custom implementation of the write-through where, somehow, every store is forwaded to the other caches). If you want your threads to access "globally" modified data, you must disable L1 and L2, but you can enable L3 (as it acts like a global-scope last-level cache). It's like early-GPUs.

It seems that the "Flush" unit is now named the "VX_cache_init" unit and is providing addresses for some kind of tag initialization at reset.

For the Vortex team, is this analysis correct? If so, do you plan to integrate a cache coherence protocol into Vortex in a near future?

Thanks.

Ifoufou, 2024-01-09:
`After receiving a snoop request, you need to distinguish whether the snoop request type is read or write`
Not an expert about snoop protocols, but I think a snoop request only intends to signal a modification on a shared data to other caches. This modification can only come subsequently after a store/write. It is then the responsibility of the snooping implementation to decide whether it performs invalidations or updates. There is no read involve, AFAIK.

`I think this operation is dangerous because dirty datas may get lost`
Why could we have dirty datas? Reset is only triggered once at the beginning of the simulation/FPGA emulation. Maybe I missed something but as I said, I think it is only used for some kind of tag initialization.

`this stupid and naive operation alone is not enough to maintain cache coherence as stated in the [paper](https://arxiv.org/abs/2110.10857).`
In the paper, by using the term "Flush" in the sentence "Flush operations among caches are provided as a means of providing weak coherent memory space", I think they meant the function "vx_flush_caches()" provided by the device interface, allowing the user to flush the whole cache hierarchy. This function was removed with the snooping protocol, which makes sense as there is no way of forwarding the invalidations.

@tinebp @hyesoon @caiyzik Could you give us some feedback? Thanks :)

closed 2026-06-14 by tinebp