vortexgpgpu/vortex #104: How to generate .hex file

JinB1, 2024-02-13:
I want to know how to create a hex file to run in Vortex.

ex)  /vortex2/tests/riscv/isa/rv32mi-p-ma_addr.hex
ex)  add assembly to add.hex

thank you

troibe, 2024-02-13:
Hi Jin,
There are two different parts to your question:
- The isa tests are from the official isa tests [here](https://github.com/riscv-software-src/riscv-tests/tree/master) and depend on [riscv-test-env](https://github.com/riscv/riscv-test-env).
- Other tests/kernels within Vortex are relying on `libvortexrt.a` like these ones [here](https://github.com/vortexgpgpu/vortex/tree/master/tests/kernel).

So both `riscv-test-env` and `libvortexrt.a` are fine to use for creating your own kernels/tests.
I would advise you to simply copy the [hello world kernel ](https://github.com/vortexgpgpu/vortex/tree/master/tests/kernel/hello)and modify it as needed.

JinB1, 2024-02-13:
Hi troibe,
Thank you for your helpful response.

closed 2024-02-13 by JinB1