vortexgpgpu/vortex #105: --app = hello is not working

JinB1, 2024-02-13:
➜  /vortex git:(master) ✗ ./ci/blackbox.sh --cores=1 --warps=1 --threads=1 --app=hello --driver=rtlsim --rebuild=1
 
CONFIGS=-DNUM_CLUSTERS=1 -DNUM_CORES=1 -DNUM_WARPS=1 -DNUM_THREADS=1    
running: CONFIGS=-DNUM_CLUSTERS=1 -DNUM_CORES=1 -DNUM_WARPS=1 -DNUM_THREADS=1     make -C ./ci/../runtime/rtlsim
running: make -C ./ci/../tests/kernel/hello run-rtlsim
make: Entering directory '/vortex/tests/kernel/hello'
../../../sim/rtlsim/rtlsim hello.bin
Running hello.bin...

I just waited for hour but it is not working ....

closed 2024-02-13 by JinB1

troibe, 2024-02-13:
Not working in which way?
You could start testing with `simx` first to see if the binary is fine and the switch to `rtlsim`.

JinB1, 2024-02-14:
In my modified vortex, project Hello was not working, So I built the original vortex and ran Hello it worked.