vortexgpgpu/vortex #107: How to get performance result  when running test/riscv/isa files

JinB1, 2024-02-14:
I am going to test each /vortex/test/riscv/isa

../../../sim/rtlsim/rtlsim -r rv32um-p-mul.hex 

and the result is 
"
Running rv32um-p-mul.hex...
Passed
"

**but I want to get a performance result like**

`➜  /vortex git:(master) ✗ ./ci/blackbox.sh --cores=1 --warps=1 --threads=1 --app=vecadd --driver=rtlsim --rebuild=1

CONFIGS=-DNUM_CLUSTERS=1 -DNUM_CORES=1 -DNUM_WARPS=1 -DNUM_THREADS=1    
running: CONFIGS=-DNUM_CLUSTERS=1 -DNUM_CORES=1 -DNUM_WARPS=1 -DNUM_THREADS=1     make -C ./ci/../runtime/rtlsim
running: make -C ./ci/../tests/opencl/vecadd run-rtlsim
make: Entering directory '/vortex/tests/opencl/vecadd'
LD_LIBRARY_PATH=/opt/pocl/runtime/lib:/vortex/runtime/rtlsim: ./vecadd -n64
Workload size=64
Create context
Allocate device buffers
Create program from kernel source
Upload source buffers
Execute the kernel
Elapsed time: 168 ms
Download destination buffer
Verify result
PASSED!
**PERF: instrs=2776, cycles=24642, IPC=0.112653
make: Leaving directory '/vortex/tests/opencl/vecadd'`**

troibe, 2024-02-15:
Hi Jin,
That performance information is a function in `libvortex` meanwhile the isa tests rely on `riscv-test-env`.
So you that function is not called.
One way to get around this would be to rebuild the tests from `riscv-tests` and wrap them in either in a call to the performance function in libvortex or have your wrappper read out the `mcycle` before and after the test.

troibe, 2024-02-15:
You might also have to check if the `rv32um-p-mul` test would even be suitable to evaluate performance.
I would suspect the test is written in a way to evaluate the correctness of the mul instruction.

JinB1, 2024-02-16:
Thank you for your response, but I'm sorry, I didn't quite understand it

I'm in a situation where I need to check both the performance and the waveform, so I've proceeded in such a way that I can view both the waveform and log files. However, I'm not sure if this is the right approach to take

in sim/rtlsim/Makefile
I added

`DEBUG = 1
ifdef DEBUG
	VL_FLAGS += --trace --trace-structs $(DBG_FLAGS)
	CXXFLAGS += -g -O0 $(DBG_FLAGS)
else    
	VL_FLAGS += -DNDEBUG
	CXXFLAGS += -O2 -DNDEBUG
endif
`
`CXXFLAGS += -DDUMP_PERF_STATS\`

and I can get the waveform and log files with using this command
`./sim/rtlsim/rtlsim -r tests/riscv/isa/rv32um-p-mul.hex > mul`

Additionally, I'm planning to convert assembly code into a binary file and then execute it. For this part, I'm considering writing the assembly code with reference to https://github.com/riscv-software-src/riscv-tests/blob/master/isa/macros/scalar/test_macros.h, converting it into a binary file, and then attempting to run it with ./sim/rtlsim/rtlsim -r ---.bin. Is this an appropriate method to follow?