vortexgpgpu/vortex #119: [Discussion] Compiler Support

danikhan632, 2024-03-27:
Really intrested with this project, I've played around with a fork of triton before that lowers to a SPIRV kernel which could interop with existing OpenCL infrastructure. Wondering if anyone here would be intrested?

troibe, 2024-07-19:
I'm assuming you are a student at Georgia Tech.
Were you able to propose your project to Prof. Hyesoon Kim?
If not I'd recommend writing up a couple of paragraphs with more details and sending her an email.

danikhan632, 2024-07-19:
yeah I discussed with prof and she said it was a good idea.
I've searched through vortex codebase quite a bit but I haven't been able to figure out if there is per cluster/SM shared memory and if there are instructions to read/write to it. Triton is very shared memory centric so a bit confused about porting it to vortex

JacobLevinson, 2024-08-06:
Read through the [docs](https://github.com/vortexgpgpu/vortex/blob/master/docs/microarchitecture.md) on the microarchitecure, this shows that the cache is shared between different levels. There is also "shared memory" support, look through the tests/regression/segmm2x for an example of that.

danikhan632, 2024-08-06:
Local memory's scope is per core, not per cluster which is pretty important for shared memory
If Vortex Cluster = Nvidia Streaming Multiproccessor(SM), then I think shared memory would have to be per cluster.
Shared memory also has its own address space and is not really a cache more like a scratchpad memory.

JacobLevinson, 2024-08-06:
> Local memory's scope is per core, not per cluster which is pretty important for shared memory If Vortex Cluster = Nvidia Streaming Multiproccessor(SM), then I think shared memory would have to be [...]

Is the Cuda SM equivalent in vortex not the "core"? What makes you say it is cluster?

danikhan632, 2024-08-06:
vortex core is more equivalant to nvidia stream processor for a few reasons:
Stream proccessors typically are 5ish stage pipeline
SMs don't have warp scedulers, stream processors do
or regsiter files, dispatch units, Load store units, ALUs etc.
Its a bit hard for me to say that a vortex core is equivalant to an SM since I don't really see too many similarites.
Below is an SM from the ampere architecture (Nvidia A100/RTX 30 series)

[image]

tinebp, 2026-06-14:
On the mapping question: a Vortex **core** maps most closely to an NVIDIA **SM** — it has the warp scheduler, the register file, the LSU/ALUs, and the per-core local memory (scratchpad) that serves as "shared memory". A **cluster** groups several cores under a shared L2. So shared memory in Vortex is the per-core local memory (addressable scratchpad, analogous to CUDA `__shared__`), which is what a Triton-style shared-memory-centric lowering would target. See `tests/regression/sgemm2x` for a shared-memory example. Closing the thread — please open a GitHub Discussion to continue the Triton-on-Vortex design.