vortexgpgpu/vortex #120: Quick demo running vecadd OpenCL kernel faield

EnkiSun86, 2024-04-10:
I cloned the vortex master branch yesterday(2024.04.09) and built it follow the README.md. The build step run properly until I run the `$ ./ci/blackbox.sh --cores=2 --app=vecadd` , it returned with an error as follows:
`Workload size=64
invalid caps id: 6
../common.mk:94: recipe for target 'run-simx' failed`
I checked the source code in the "runtime/simx/vortex.cpp" and found that it returns the error since  there is no "VX_CAPS_LOCAL_MEM_SIZE" caps support in the simx runtime. And I found that the pocl runtime lib "libpocl-devices-vortex.so" exist a "vx_dev_caps" call in the newest library download by the "ci/toolchain_install.sh" script. I wonder is there any mismatching between the simx runtime and the pocl runtime?

blindguy789, 2024-04-14:
have you found a solution?

Geunyoung00, 2024-05-13:
I got the same error running in simx.

I have tested in fpga, $ TARGET=fpga ./ci/blackbox.sh --driver=opae --app=vecadd
And I had two errors: 
1. [VXDRV] Error: 'api.fpgaPropertiesGetLocalMemorySize(filter, &device->global_mem_size)' returned 6, not supported!
2. OpenCL Error: 'clGetDeviceIDs(platform_id, CL_DEVICE_TYPE_DEFAULT, 1, &device_id, NULL)' returned -33!

tinebp, 2024-08-02:
should be fixed now

closed 2024-08-02 by tinebp