vortexgpgpu/vortex #122: FPU Errors When Testing Vortex on Intel Devcloud

HeeeonLee, 2024-04-18:
### Q : Any alternative for Intel's DSP Builder Library?

Here is why it is necessary to execute FPU operations in Intel's Devcloud(Arria 10)
I've managed to upload vortex to Devcloud, but here is an issue. I will my debugging process one by one

### 1. Errors on acl_fmadd, acl_fdiv, etc
This is /home/uxxxxxx/vortex/hw/syn/altera/opae/build32_arria10_fpga_1c/synth/build.log 
You can see the error regarding acl_fmadd, acl_fdiv, etc. popped up.
[image]

### 2. It led me to VX_fpu_fma.sv
so I followed the error to /home/u220123/vortex/hw/syn/altera/opae/build32_arria10_fpga_1c/src/VX_fpu_fma.sv
[image]

### 3. ip_gen.sh checking
I checked ip_gen,log. It seems like dspba is not present.
[image]

### 4. Intel's DSP Builder Library
I figured that DSP Builder Library is needed. It's a bit expensive so I'm trying to find alternatives. Any suggestion?

HeeeonLee, 2024-04-18:
+ When FPU is disabled, FPGA works well.

HeeeonLee, 2024-04-23:
I found the solution. just set Quartus Pro or Quartus Standard when executing "tools_setup".
[image]

and then, add following path
export DEVICE_FAMILY="Arria10"

closed 2024-04-23 by HeeeonLee

msaideroglu, 2024-10-09:
@heeeoneee Hi. This is not a real solution. `export DEVICE_FAMILY="Arria10"` makes pass that part because you're not giving correct define parameters as in Makefile script that 
```
ifeq ($(DEVICE_FAMILY), stratix10)
    CONFIGS += -DALTERA_S10
endif
ifeq ($(DEVICE_FAMILY), arria10)
    CONFIGS += -DALTERA_A10
endif
```
Did you find any other solution to this?

> I found the solution. just set Quartus Pro or Quartus Standard when executing "tools_setup". ![Screenshot 2024-04-23 at 3 46 34 PM](https://private-user-images.githubusercontent.com/129277649/324723839-e65272c2-8d89-45b9-94cf-919e2c21bc97.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg0Nzk3MjAsIm5iZiI6MTcyODQ3OTQyMCwicGF0aCI6Ii8xMjkyNzc2NDkvMzI0NzIzODM5LWU2NTI3MmMyLThkODktNDViOS05NGNmLTkxOWUyYzIxYmM5Ny5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAwOVQxMzEwMjBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lYjZkYjkyMDdiY2M2OTlhYjY0ZGVjNDQ0NDljNDZiOTc4M2E5Y2Q1ODEwY2IzY2QxMjI5ZmUwYWQ1ZjM2MjMzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.V_C2Ro3g8xhqlOB0cnrQ_5_z7jNJQoWg5LtYCNkuQNU)
> 
> and then, add following path export DEVICE_FAMILY="Arria10"