vortexgpgpu/vortex #126: way_idx may not correct

linuxbest, 2024-05-13:
With the default setting, the NUM_WAYS of icache is 1, so the way_idx should always be 0, but it will change to 1. This behavior is acceptable with Verilator but fails with other simulators.

`
    wire [NUM_WAYS-1:0][`CS_WORD_WIDTH-1:0] per_way_rdata;

    if (`CS_WORDS_PER_LINE > 1) begin
        assign per_way_rdata = rdata[wsel];
    end else begin
        `UNUSED_VAR (wsel)
        assign per_way_rdata = rdata;
    end    

    assign read_data = per_way_rdata[way_idx];
`

Verilator:
[image]

VCS:
[image]

```
                 597: -bank0 tag-fill: addr=0x80000000, way=1, blk_addr=0, tag_id=0x20000.              <<<< way=1, ?
                 599: -bank0 mshr-dequeue: addr=0x80000000, id=0 (#0)
                 599: -bank0 mshr-pop: addr=0x80000000, tag=0x0, req_idx=0 (#0)
                 599: -bank0 data-fill: addr=0x80000000, way=1, blk_addr=0, data=0x2800bfff002930062900b1203031300000317fc1022f313c000ef0002800b001002930ec000ef0002800bfff002930062900b1343031300000317fc1022f3
                 601: -bank0 mshr-table
                 601: -bank0 tag-hit: addr=0x80000000, way=0, blk_addr=0, tag_id=0x20000 (#0)
                 603: -bank0 core-rd-rsp: addr=0x80000000, tag=0x0, req_idx=0, data=0xfc1022f3 (#0)
                 603: -bank0 data-read: addr=0x80000000, way=1, blk_addr=0, data=0xfc1022f3 (#0). <<<<<< way=1, it should be 0 when NUM_WAY=1
                 605: I$0 rsp: wid=0, PC=0x80000000, tmask=0001, instr=0xfc1022f3 (#0)
```

linuxbest, 2024-05-13:
It looks bug with hot_encoder, thanks.

```
diff --git a/hw/rtl/libs/VX_onehot_encoder.sv b/hw/rtl/libs/VX_onehot_encoder.sv
index 92c7d1ea..adaa4183 100644
--- a/hw/rtl/libs/VX_onehot_encoder.sv
+++ b/hw/rtl/libs/VX_onehot_encoder.sv
@@ -29,7 +29,7 @@ module VX_onehot_encoder #(
 ); 
     if (N == 1) begin
 
-        assign data_out  = data_in;
+        assign data_out  = 0;
         assign valid_out = data_in;
 
     end else if (N == 2) begin
```

linuxbest, 2024-05-14:
The onehot encoder looks wrong to me.

When N==1, the encoder output data should be always 0.

Thanks.

```
diff --git a/hw/rtl/libs/VX_onehot_encoder.sv b/hw/rtl/libs/VX_onehot_encoder.sv
index 92c7d1ea..adaa4183 100644
--- a/hw/rtl/libs/VX_onehot_encoder.sv
+++ b/hw/rtl/libs/VX_onehot_encoder.sv
@@ -29,7 +29,7 @@ module VX_onehot_encoder #(
 ); 
     if (N == 1) begin
 
-        assign data_out  = data_in;
+        assign data_out  = 0;
         assign valid_out = data_in;
 
     end else if (N == 2) begin
```

tinebp, 2024-08-02:
Good catch!
We would love to have you join our development effort, we are looking for hardware developers who can contribute to this open-source effort. Please join our discussion mailing list, we host monthly dev meetings
https://vortex.cc.gatech.edu/community

tinebp, 2024-08-02:
https://github.com/vortexgpgpu/vortex/commit/3075c1737bf86dfb4a1375ebef84a46492990050

closed 2024-08-02 by tinebp