vortexgpgpu/vortex #128: Xilinx Alveo U200 support

RaulMurillo, 2024-05-15:
While the [README.md](https://github.com/vortexgpgpu/vortex/blob/15ca8290d0c3cfcc559e78cec98d87172c7e6719/README.md?plain=1#L20) mentions U250 as a supported FPGA, it seems Alveo U200 is supported at [runtime](https://github.com/vortexgpgpu/vortex/blob/15ca8290d0c3cfcc559e78cec98d87172c7e6719/runtime/xrt/vortex.cpp#L74).
Also, no [build examples](https://github.com/vortexgpgpu/vortex/blob/15ca8290d0c3cfcc559e78cec98d87172c7e6719/hw/syn/xilinx/README#L17) are shown for either U200 or U250.

I was wondering therefore which of the two cards (U200 or U250) is supported, and if it would be enough to build with the option `PLATFORM=xilinx_u200_gen3x16_xdma_1_202211_1` or similar. If not, what modifications would be necessary to run Vortex on this FPGA? Maybe this is just a documentation issue.

tinebp, 2024-05-15:
modifying this document is a place to start
https://github.com/vortexgpgpu/vortex/blob/15ca8290d0c3cfcc559e78cec98d87172c7e6719/runtime/xrt/vortex.cpp#L71

tinebp, 2026-06-14:
Resolved in the current tree. The runtime no longer hardcodes a per-card platform table (the U200/U250 list you referenced is gone). `sw/runtime/xrt/vortex.cpp` loads the xclbin from `XRT_XCLBIN_PATH` generically and derives memory banks from the platform, so U200 is targeted simply by building with `PLATFORM=xilinx_u200_gen3x16_xdma_...` and running against that bitstream — no source edit needed. Closing.

closed 2026-06-14 by tinebp