vortexgpgpu/vortex #140: IPC mistach.

yliu921, 2024-06-16:
Hello,

Thanks for this wonderful project.
I'm trying to deploy the Vortex GPU to Xilinx Alveo U50.
However, the IPC mismatches the results in Vortex Paper. 
[image]
This is the measurement. 
[image]

I used the default configuration to implement the FPGA bitstream:

> PREFIX=build_base_1c NUM_CORES=1 TARGET=hw PLATFORM=xilinx_u50_gen3x16_xdma_5_202210_1 make > build_u50_hw_base_1c.log 2>&1 &

The workload is set to:
> vecadd with 2048

May I ask the configuration that can reproduce the results?

Thanks!

Ifoufou, 2024-06-18:
Hi @yliu921, just curious: did you try with caches enabled?

hyesoon, 2024-06-21:
vortex paper (MICRO) is with vortex 1.0 and currently we are at vortex 2.1 .

closed 2024-07-01 by yliu921