vortexgpgpu/vortex #143: Self build toolchains for RISC-V lacking __libc_fini_array entry

EnkiSun86, 2024-06-24:
I'm trying to bootstrap and compile the toolchains for Vortex.  I followed the README.vortex in the pocl source file pulled from https://github.com/vortexgpgpu/pocl/tree/vortex .
```
#### Compiling pocl on RiscV ####

## Dependencies:

- sudo apt-get -y install \
  binutils build-essential libtool texinfo \
  gzip zip unzip patchutils curl git \
  make cmake ninja-build automake bison flex gperf \
  grep sed gawk python bc \
  zlib1g-dev libexpat1-dev libmpc-dev \
  libglib2.0-dev libfdt-dev libpixman-1-dev

## Setting tools directory

export TOOLDIR=$HOME/tools

## Building RiscV GNU Toolchain (gcc, binutils, etc..)

- git clone --depth=1 --recursive https://github.com/riscv-collab/riscv-gnu-toolchain.git
- mkdir build && cd build
- export CPATH=$TOOLDIR/GNU/include
- export LIBRARY_PATH=$TOOLDIR/GNU/lib
- ../configure --prefix=$TOOLDIR/riscv32-gnu-toolchain --with-cmodel=medany --with-arch=rv32imf --with-abi=ilp32f
#../configure --prefix=$TOOLDIR/riscv64-gnu-toolchain --with-cmodel=medany --with-arch=rv64imafd --with-abi=lp64d
- make -j`nproc`
- make -j`nproc` build-qemu

## Building LLVM for Vortex

- git clone --recursive --branch vortex https://github.com/vortexgpgpu/llvm.git
- cd llvm
- mkdir build && cd build
- export LLVM_PREFIX=$TOOLDIR/llvm-vortex
- export RISCV_TOOLCHAIN_PATH=$TOOLDIR/riscv64-gnu-toolchain
- cmake -G "Unix Makefiles" -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=$LLVM_PREFIX -DLLVM_ENABLE_PROJECTS="clang;lld" -DBUILD_SHARED_LIBS=True -DLLVM_TARGETS_TO_BUILD="RISCV" -DLLVM_ABI_BREAKING_CHECKS=FORCE_OFF -DLLVM_INCLUDE_BENCHMARKS=OFF -DLLVM_INCLUDE_EXAMPLES=OFF -DLLVM_INCLUDE_TESTS=OFF -DDEFAULT_SYSROOT=$RISCV_TOOLCHAIN_PATH/riscv32-unknown-elf -DLLVM_DEFAULT_TARGET_TRIPLE="riscv32-unknown-elf" ../llvm
- make -j`nproc`
- make install

## Sanity test your new RISC-V LLVM

- echo -e '#include <stdio.h>\n int main(void) { printf("Hello world!\\n"); return 0; }' > hello.c
- clang hello.c
- qemu-riscv32 hello

## Building llvm-spirv
- git clone --depth=1 -b release/10.x  https://github.com/KhronosGroup/SPIRV-LLVM-Translator.git
- cd llvm-spirv
- mkdir build && cd build
- export LLVM_VORTEX=$TOOLDIR/llvm-vortex
- cmake .. -DLLVM_DIR=$LLVM_VORTEX -DCMAKE_INSTALL_PREFIX=$LLVM_VORTEX
- make llvm-spirv -j`nproc`
- make install
# manually copy over llvm-spirv binary
- cp ./tools/llvm-spirv/llvm-spirv $LLVM_VORTEX/bin

## build POCL compiler
- git clone --branch vortex --recursive https://github.com/vortexgpgpu/pocl
- cd pocl
- mkdir build_cc && cd build_cc
- export POCL_CC_PATH=$TOOLDIR/pocl/compiler
- export LLVM_VORTEX=$TOOLDIR/llvm-vortex
- cmake -G "Unix Makefiles" -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=$POCL_CC_PATH -DOCS_AVAILABLE=ON -DWITH_LLVM_CONFIG=$LLVM_VORTEX/bin/llvm-config -DENABLE_VORTEX=ON -DBUILD_TESTS=OFF -DPOCL_DEBUG_MESSAGES=ON -DENABLE_ICD=OFF -DCLANG_MARCH_FLAG:STRING=-mcpu= -DLLC_HOST_CPU=generic-rv32 ..
- make -j`nproc`
- make install

## build POCL runtime
- git clone --branch vortex --recursive https://github.com/vortexgpgpu/pocl
- cd pocl
[... 134 lines cut ...]
```
This led to the build of vx_syscalls.c lack of  entry of __libc_init_array and  __libc_fini_array.  I fixed this by this change:
```
diff --git a/kernel/src/vx_syscalls.c b/kernel/src/vx_syscalls.c
index 6ff9fbb..ad0777b 100644
--- a/kernel/src/vx_syscalls.c
+++ b/kernel/src/vx_syscalls.c
@@ -65,7 +65,7 @@ void __init_tls(void) {
   memset(__thread_self + (size_t)__tbss_offset, 0, (size_t)__tbss_size);
 }
 
-#ifdef HAVE_INITFINI_ARRAY
+#ifdef _HAVE_INITFINI_ARRAY
 
 // These magic symbols are provided by the linker.
 extern void (*__preinit_array_start []) (void) __attribute__((weak));
@@ -96,7 +96,7 @@ void __libc_init_array (void) {
 }
 #endif
 
-#ifdef HAVE_INITFINI_ARRAY
+#ifdef _HAVE_INITFINI_ARRAY
 extern void (*__fini_array_start []) (void) __attribute__((weak));
 extern void (*__fini_array_end []) (void) __attribute__((weak));
```

EnkiSun86, 2024-06-24:
The remaining puzzle is how to bootstrap build this libc32 myself.

EnkiSun86, 2024-06-25:
I fixed the `ld.lld: warning: cannot find entry symbol _start; not setting start address` by linking the crt1.o in libc32 manually. However, the qemu-riscv32 still can not run the a.out since there are some Vortex-specific instructions in the a.out like `vx_spilt`. More effort is needed to modify the qemu to support the evaluation of Vortex

tinebp, 2026-06-14:
The current tree ships a complete from-source toolchain guide: `docs/building_toolchain.md`. It covers the RISC-V GNU toolchain, LLVM-Vortex, compiler-rt, and **musl libc** (the toolchain moved to musl, which removes the old newlib `__libc_fini_array`/`crt` bootstrap problem you hit), plus POCL. Closing — please follow that document.

closed 2026-06-14 by tinebp