vortexgpgpu/vortex #156: Ramulator 2.0 Update Issue w/gcc version on Ubuntu 22.04

JacobLevinson, 2024-07-22:
Hello. I just recloned the latest vortex and when I try to make for the first time (after following standard setup procedure with **prebuilt toolchain**, I get an error that ranges from #include <ranges> is not found in Ramulator build (likely due to compiler version). Default gcc/g++ version on ubuntu 22.04 is 9.4.0, which does not support <ranges>. 

This can be fixed by installing gcc/g++ 13 from the ubuntu-toolchain-r/test ppa:

sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt update
sudo apt install gcc-13 g++-13
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-13 100
sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-13 100

jacob@DESKTOP-TL8HK1P:~/vortex/build$ make -s
-- The CXX compiler identification is GNU 9.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Configuring Ramulator ...
-- Build type: Release
Configuring yaml-cpp...
Done configuring yaml-cpp.
Configuring spdlog...
-- Build spdlog: 1.11.0
-- Looking for C++ include pthread.h
-- Looking for C++ include pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Build type: Release
Done configuring spdlog.
Configuring argparse...
Done configuring argparse.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jacob/vortex/third_party/ramulator/build
Scanning dependencies of target spdlog
Scanning dependencies of target ramulator-controller
Scanning dependencies of target yaml-cpp
Scanning dependencies of target ramulator-base
[  2%] Building CXX object _deps/spdlog-build/CMakeFiles/spdlog.dir/src/spdlog.cpp.o
[  2%] Building CXX object _deps/yaml-cpp-build/CMakeFiles/yaml-cpp.dir/src/binary.cpp.o
[  3%] Building CXX object src/base/CMakeFiles/ramulator-base.dir/factory.cpp.o
[  4%] Building CXX object src/dram_controller/CMakeFiles/ramulator-controller.dir/impl/bh_dram_controller.cpp.o
In file included from /home/jacob/vortex/third_party/ramulator/src/dram/dram.h:10,
                 from /home/jacob/vortex/third_party/ramulator/src/dram_controller/bh_controller.h:11,
                 from /home/jacob/vortex/third_party/ramulator/src/dram_controller/impl/bh_dram_controller.cpp:1:
/home/jacob/vortex/third_party/ramulator/src/dram/spec.h:9:10: fatal error: ranges: No such file or directory
    9 | #include <ranges>
      |          ^~~~~~~~
compilation terminated.
make[4]: *** [src/dram_controller/CMakeFiles/ramulator-controller.dir/build.make:63: src/dram_controller/CMakeFiles/ramulator-controller.dir/impl/bh_dram_controller.cpp.o] Error 1
make[3]: *** [CMakeFiles/Makefile2:1418: src/dram_controller/CMakeFiles/ramulator-controller.dir/all] Error 2
make[3]: *** Waiting for unfinished jobs....
.
.
.
.
.
make[2]: *** [Makefile:152: all] Error 2
make[1]: *** [Makefile:9: ramulator] Error 2
make: *** [Makefile:4: all] Error 2

liumengbjut, 2024-07-26:
the same question

tinebp, 2024-08-02:
The Vortex toolchain setup works, we enforce that commits to the master branch are tested in the continuous integration.
These recent changes introducing new Verilator and Ramulator 2.0 require a clean git clone since the toolchain and dependencies have changed.
A robust way to simplify your setup process with future Vortex changes is to follow the steps we use for the CI testing:
$ sudo bash ./ci/system_updates.sh
$ ci/toolchain_install.sh --all

closed 2024-08-02 by tinebp