vortexgpgpu/vortex #157: Additional Issue with New Update - Verilator "Inconsistent vortex home"

JacobLevinson, 2024-07-22:
After fixing the ramulator issue in issue #156 I get another error, this time with verilator:

%Error: verilator: VERILATOR_ROOT is set to inconsistent path. Suggest leaving it unset.
%Error: VERILATOR_ROOT=/home/jacob/tools/verilator
make[2]: *** [Makefile:88: /home/jacob/vortex/build/sim/rtlsim/rtlsim] Error 1
make[1]: *** [Makefile:6: all] Error 2
make: *** [Makefile:6: all] Error 2

msaideroglu, 2024-07-23:
I'm getting same error after last commit and update on prebuilt toolchain.

diwakar1998, 2024-07-29:
I am having the same issue when i tried to setup vortex in a  docker container.

tinebp, 2024-08-02:
This is an expected change that Verilator 5.0 introduced.
On a clean enlistment, you should not have to worry about this since we have updated toolchain_env.sh
If you are using ~/.bashrc, make sure that it directly sources toolchain_env.sh from your build folder, do not hardcode any environment. 
Read the comments on other related issues I closed for how to avoid not having to face this type of issue in the future. update.

closed 2024-08-02 by tinebp