vortexgpgpu/vortex #160: Xilinx Synthesis Fails

msaideroglu, 2024-07-26:
I used the automated make script to generate a Vivado project for FPGA implementation of Vortex. It creates the project however gives error during synthesis:

> [Synth 8-2671] single value range is not allowed in this mode of verilog ["/vortex/build/hw/syn/xilinx/test/project_1/src/Vortex_top.v":51]

When I reconfig the file type as SystemVerilog, then behavioral simulation gives this error during elaboration phase:

> [VRFC 10-2063] Module <design_1_axi_bram_ctrl_0_0> not found while processing module instance <axi_bram_ctrl_0> ["/vortex/build/hw/syn/xilinx/test/project_1/project_1.ip_user_files/bd/design_1/sim/design_1.v":131]

xpww, 2024-08-03:
I also have this issue, Vortex_top.v uses the syntax of sv, causing the syntax to emulation normally.

redpanda3, 2024-08-24:
Hi, may I ask how long it would take for synthesis?