vortexgpgpu/vortex #162: Issue building vortex on the newest updates: Verilator

JacobLevinson, 2024-07-29:
For the pasdt few weeks, I have been unable to make -s with a fresh vortex clone. I get the following:

jacob@DESKTOP-TL8HK1P:~/vortex/build$ make -s
make[3]: execvp: ccache: Permission denied
make[3]: *** [VVortex.mk:72: float_dpi.o] Error 127
%Error: make -C /home/jacob/vortex/build/sim/rtlsim/rtlsim.obj_dir -f VVortex.mk -j 16 exited with 2
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /home/jacob/tools/verilator/share/verilator/bin/verilator_bin --build --exe --language 1800-2009 --assert -Wall -Wpedantic -Wno-DECLFILENAME -Wno-REDEFMACRO -Wno-GENUNNAMED --x-initial unique --x-assign unique /home/jacob/vortex/sim/rtlsim/verilator.vlt -DSIMULATION -DSV_DPI -DXLEN_32 -I/home/jacob/vortex/hw/rtl -I/home/jacob/vortex/hw/dpi -I/home/jacob/vortex/hw/rtl/libs -I/home/jacob/vortex/hw/rtl/interfaces -I/home/jacob/vortex/hw/rtl/core -I/home/jacob/vortex/hw/rtl/mem -I/home/jacob/vortex/hw/rtl/cache -I/home/jacob/vortex/hw/rtl/fpu /home/jacob/vortex/hw/rtl/VX_gpu_pkg.sv /home/jacob/vortex/hw/rtl/fpu/VX_fpu_pkg.sv /home/jacob/vortex/hw/rtl/core/VX_trace_pkg.sv --cc Vortex --top-module Vortex -j 16 -DNDEBUG /home/jacob/vortex/sim/common/util.cpp /home/jacob/vortex/sim/common/mem.cpp /home/jacob/vortex/sim/common/rvfloats.cpp /home/jacob/vortex/sim/common/dram_sim.cpp /home/jacob/vortex/hw/dpi/util_dpi.cpp /home/jacob/vortex/hw/dpi/float_dpi.cpp /home/jacob/vortex/sim/rtlsim/processor.cpp /home/jacob/vortex/sim/rtlsim/main.cpp -CFLAGS -std=c++17\ -Wall\ -Wextra\ -Wfatal-errors\ -Wno-array-bounds\ -fPIC\ -Wno-maybe-uninitialized\ -I/home/jacob/vortex/build/hw\ -I/home/jacob/vortex/sim/common\ -I/home/jacob/vortex/third_party/softfloat/source/include\ -I/home/jacob/vortex/third_party/ramulator/ext/spdlog/include\ -I/home/jacob/vortex/third_party/ramulator/ext/yaml-cpp/include\ -I/home/jacob/vortex/third_party/ramulator/src\ -DXLEN_32\ \ -O2\ -DNDEBUG\ -DSTARTUP_ADDR=0x80000000 -LDFLAGS /home/jacob/vortex/third_party/softfloat/build/Linux-x86_64-GCC/softfloat.a\ -Wl\,-rpath\,/home/jacob/vortex/third_party/ramulator\ \ -L/home/jacob/vortex/third_party/ramulator\ -lramulator --Mdir /home/jacob/vortex/build/sim/rtlsim/rtlsim.obj_dir -o /home/jacob/vortex/build/sim/rtlsim/rtlsim
make[2]: *** [Makefile:88: /home/jacob/vortex/build/sim/rtlsim/rtlsim] Error 2
make[1]: *** [Makefile:6: all] Error 2
make: *** [Makefile:8: all] Error 2

And when I sudo make -s:
make[2]: verilator: Command not found
make[2]: *** [Makefile:88: /home/jacob/vortex/build/sim/rtlsim/rtlsim] Error 127
make[1]: *** [Makefile:6: all] Error 2
make: *** [Makefile:8: all] Error 2

This has made testing and developing with vortex impossible, so if a fix could be pushed it would be much appreciated. Thank you.

fadialzammar, 2024-07-31:
Does `which verilator` return anything? It looks like you don't have it in your path. Have you sourced the `build/ci/toolchain_env.sh` file?

JacobLevinson, 2024-08-01:
Yes to both:

jacob@DESKTOP-TL8HK1P:~/vortex/build$ source ./ci/toolchain_env.sh
jacob@DESKTOP-TL8HK1P:~/vortex/build$ make -s
make[3]: execvp: ccache: Permission denied
make[3]: *** [VVortex.mk:72: float_dpi.o] Error 127
%Error: make -C /home/jacob/vortex/build/sim/rtlsim/rtlsim.obj_dir -f VVortex.mk -j 16 exited with 2
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /home/jacob/tools/verilator/share/verilator/bin/verilator_bin --build --exe --language 1800-2009 --assert -Wall -Wpedantic -Wno-DECLFILENAME -Wno-REDEFMACRO -Wno-GENUNNAMED --x-initial unique --x-assign unique /home/jacob/vortex/sim/rtlsim/verilator.vlt -DSIMULATION -DSV_DPI -DXLEN_32 -I/home/jacob/vortex/hw/rtl -I/home/jacob/vortex/hw/dpi -I/home/jacob/vortex/hw/rtl/libs -I/home/jacob/vortex/hw/rtl/interfaces -I/home/jacob/vortex/hw/rtl/core -I/home/jacob/vortex/hw/rtl/mem -I/home/jacob/vortex/hw/rtl/cache -I/home/jacob/vortex/hw/rtl/fpu /home/jacob/vortex/hw/rtl/VX_gpu_pkg.sv /home/jacob/vortex/hw/rtl/fpu/VX_fpu_pkg.sv /home/jacob/vortex/hw/rtl/core/VX_trace_pkg.sv --cc Vortex --top-module Vortex -j 16 -DNDEBUG /home/jacob/vortex/sim/common/util.cpp /home/jacob/vortex/sim/common/mem.cpp /home/jacob/vortex/sim/common/rvfloats.cpp /home/jacob/vortex/sim/common/dram_sim.cpp /home/jacob/vortex/hw/dpi/util_dpi.cpp /home/jacob/vortex/hw/dpi/float_dpi.cpp /home/jacob/vortex/sim/rtlsim/processor.cpp /home/jacob/vortex/sim/rtlsim/main.cpp -CFLAGS -std=c++17\ -Wall\ -Wextra\ -Wfatal-errors\ -Wno-array-bounds\ -fPIC\ -Wno-maybe-uninitialized\ -I/home/jacob/vortex/build/hw\ -I/home/jacob/vortex/sim/common\ -I/home/jacob/vortex/third_party/softfloat/source/include\ -I/home/jacob/vortex/third_party/ramulator/ext/spdlog/include\ -I/home/jacob/vortex/third_party/ramulator/ext/yaml-cpp/include\ -I/home/jacob/vortex/third_party/ramulator/src\ -DXLEN_32\ \ -O2\ -DNDEBUG\ -DSTARTUP_ADDR=0x80000000 -LDFLAGS /home/jacob/vortex/third_party/softfloat/build/Linux-x86_64-GCC/softfloat.a\ -Wl\,-rpath\,/home/jacob/vortex/third_party/ramulator\ \ -L/home/jacob/vortex/third_party/ramulator\ -lramulator --Mdir /home/jacob/vortex/build/sim/rtlsim/rtlsim.obj_dir -o /home/jacob/vortex/build/sim/rtlsim/rtlsim
make[2]: *** [Makefile:88: /home/jacob/vortex/build/sim/rtlsim/rtlsim] Error 2
make[1]: *** [Makefile:6: all] Error 2
make: *** [Makefile:8: all] Error 2
jacob@DESKTOP-TL8HK1P:~/vortex/build$ sudo make -s
[sudo] password for jacob: 
make[2]: verilator: Command not found
make[2]: *** [Makefile:88: /home/jacob/vortex/build/sim/rtlsim/rtlsim] Error 127
make[1]: *** [Makefile:6: all] Error 2
make: *** [Makefile:8: all] Error 2
jacob@DESKTOP-TL8HK1P:~/vortex/build$ which verilator
/home/jacob/tools/verilator/bin/verilator
jacob@DESKTOP-TL8HK1P:~/vortex/build$

tinebp, 2024-08-02:
Vortex now uses Verilator 5.0.
If you update your system, we recommend a clean install with a git clone starting from scratch.
the toolchain will deeply all new components as well.
A good tip for resolving setup issues is to always refer to the CI regression setup
vortex/.github/workflows/ci.yml
It runs the following setup steps on a clean repository.
$ sudo bash ./ci/system_updates.sh
$ ci/toolchain_install.sh --all

closed 2024-08-02 by tinebp