vortexgpgpu/vortex #172: About Stratix10

demonk69, 2024-08-19:
Hello, I am a beginner and while trying to deploy Vortex on the Intel Stratix10 development board, I noticed that Intel mentioned in the Intel Acceleration Stack Quick Start Guide: Intel FPGA Programmable Acceleration Card D5005 that when using Stratix10 as an FPGA Programmable Acceleration Card, it requires support from the RHEL system. But Vortex's platform OS only has Ubuntu and CentOS. Can I deploy Vortex to Stratix10 on Ubuntu 20.04？

yli147, 2024-10-30:
Have you any progress on it ?

demonk69, 2024-10-30:
> Have you any progress on it ?

Not yet

yli147, 2024-10-30:
Actually I can build the image S10，  but I'm suffering a lot on my D5005 card， the first issue is thermal， the other is fw version. By the way how did you access to the devcloud ?

I'm using ubuntu 16.04, I can install the d5005 sdk on it.

Download d5005_pac_ias_2_0_1_pv_rte_installer.tar.gz from https://www.intel.com/content/www/us/en/software-kit/665814/intel-fpga-pac-d5005-acceleration-stacks-v2-0-1.html
tar xvf d5005_pac_ias_2_0_1_pv_rte_installer.tar.gz
cd d5005_pac_ias_2_0_1_pv_rte_installer
change "is_ubuntu" to 1 in setup.sh
./setup.sh
Accept the license.
Skip the OPAE
source /home/<username>/inteldevstack/init_env.sh

cd ~/intelrtestack/d5005_ias_2_0_1_b237/sw/opae-1.1.4-8
mkdir build
cd build
cmake -DCPACK_GENERATOR=DEB ..
make
Fix the issue in d5005_ias_2_0_1_b237/sw/opae-1.1.4-8/tools/extra/pac/fpgabist/dma/fpga_dma.cpp

make package_deb
sudo dpkg -i  opae*.deb