vortexgpgpu/vortex #190: Intel Opae problems in Devcloud

msaideroglu, 2024-10-10:
PREFIX=test1 TARGET=fpga NUM_CORES=4 make gives

```
text macro "PLATFORM_MEMORY_ADDR_WIDTH" is undefined File: local_mem_cfg_pkg.sv Line: 49 
text macro "PLATFORM_MEMORY_BURST_CNT_WIDTH" is undefined File: local_mem_cfg_pkg.sv Line: 52   
object "LOCAL_MEM_DATA_WIDTH" is not declared File: local_mem_cfg_pkg.sv Line: 55
```

and goes on.

When I add below part to Makefile

```
#ifeq ($(TARGET), fpga)
# CFLAGS += -DPLATFORM_PROVIDES_LOCAL_MEMORY -DPLATFORM_MEMORY_BANKS=1 -DPLATFORM_MEMORY_ADDR_WIDTH=32 -DPLATFORM_MEMORY_DATA_WIDTH=512 -DPLATFORM_MEMORY_BURST_CNT_WIDTH=4 -DNOPAE -DSV_DPI
#endif
```

It gives `ISSUE_RATIO is not declared. `

msaideroglu, 2024-10-11:
Update: I tried to clone and build a commit which is one month older in v2.2. It doesn't give  
`text macro "PLATFORM_MEMORY_ADDR_WIDTH" is undefined File: local_mem_cfg_pkg.sv Line: 49`
it complains about not to find acl_fmadd module etc. I found a folder altera from older releases and put in fpu folder and added Makefile:
--`FPU_INCLUDE = -I$(RTL_DIR)/fpu`
++` FPU_INCLUDE = -I$(RTL_DIR)/fpu -I$(RTL_DIR)/fpu/altera/$(DEVICE_FAMILY)`
Finally, it sucessfully generates vortex_afu.gbs with 

> ===========================================================================
 PR AFU compilation complete
 AFU gbs file is 'vortex_afu.gbs'
 Design meets timing
===========================================================================

However, now when I try `TARGET=fpga ./ci/blackbox.sh --driver=opae --app=sgemm --args="-n128"`
it gives 

```
~/Desktop/GPGPU/Vortex_GPGPU/vortex_v2.2/build$ TARGET=fpga ./ci/blackbox.sh --driver=opae --app=sgemm --args="-n128"
VORTEX_DRIVER=opae ./sgemm -n128
Workload size=128
[VXDRV] Error: 'api_.fpgaPropertiesGetLocalMemorySize(filter, &global_mem_size_)' returned 6, not supported!
Create context
Create program from kernel source
Upload source buffers
Execute the kernel
Elapsed time: 26 ms
Download destination buffer
Verify result
PASSED!
PERF: core0: instrs=3898993, cycles=4637528, IPC=0.840748
PERF: core1: instrs=3898993, cycles=4154493, IPC=0.938500
PERF: core2: instrs=3898993, cycles=4615820, IPC=0.844702
PERF: core3: instrs=3898993, cycles=4385463, IPC=0.889072
PERF: instrs=15595972, cycles=4637528, IPC=3.362993
```
How can I resolve `[VXDRV] Error: 'api_.fpgaPropertiesGetLocalMemorySize(filter, &global_mem_size_)' returned 6, not supported!` error? @tinebp

tinebp, 2024-10-11:
fpgaPropertiesGetLocalMemorySize error can be ignored, the API is not supported in that opae version!
PLATFORM_MEMORY_ADDR_WIDTH should be declared in the FPGA platform environment.
Which FPGA are you using?

msaideroglu, 2024-10-11:
Hi @tinebp. Thanks for the answer. I'm using arria10 in DevCloud.

msaideroglu, 2024-10-14:
I'm getting `out of local memory: needed= 32 bytes, available= 0 bytes` error during some of my OpenCL test code runs with opae. Is it because of Local Memory disabled? If yes, how can I config it? @tinebp

yli147, 2024-10-26:
I'm also getting the error 
[image]

msaideroglu, 2024-10-26:
@yli147 it seems that one of the last commits deleted **_VX_gpr_slice.sv_**. So if you clone and reinstall Vortex, the problem will go away. However I don't know why issues are not being responded by developers. It's helpful for them to debug some critical problems and easily fix them.

yli147, 2024-10-30:
> @yli147 it seems that one of the last commits deleted **_VX_gpr_slice.sv_**. So if you clone and reinstall Vortex, the problem will go away. However I don't know why issues are not being responded by developers. It's helpful for them to debug some critical problems and easily fix them.

Thanks, after reverting the commit, I can generated the image successfully for Altera.  by the way, which Altera FPGA are you using ?  Currently I have a https://www.intel.com/content/www/us/en/products/details/fpga/development-kits/stratix/10-gx.html and and old https://www.intel.com/content/www/us/en/products/sku/193921/intel-fpga-pac-d5005/specifications.html. Not sure which one are you using for testing on FPGA. Thanks

msaideroglu, 2024-10-30:
> Thanks, after reverting the commit, I can generate the image successfully for Altera. By the way, which Altera FPGA are you using ? Currently I have a https://www.intel.com/content/www/us/en/products/details/fpga/development-kits/stratix/10-gx.html and and old https://www.intel.com/content/www/us/en/products/sku/193921/intel-fpga-pac-d5005/specifications.html. Not sure which one are you using for testing on FPGA. Thanks

I used Intel DevCloud for FPGA. They had Arria and Atratix 10 PAC cards instead of DevKits. I'm also wondering that if I can use a Stratix 10 DevKit for Vortex.

yli147, 2024-10-30:
This is cool, do you know how to apply access for the DevCloud for FPGA ?  Thanks

msaideroglu, 2024-10-30:
> This is cool, do you know how to apply access for the DevCloud for FPGA ? Thanks

you can have a look at [https://devcloud.intel.com/oneapi/documentation/connect-with-ssh-linux-macos/]()

tinebp, 2024-11-14:
https://github.com/vortexgpgpu/vortex/commit/8230b37411dfe28fe1b59a25a5de4c7de276cf90

yli147, 2024-11-15:
> https://devcloud.intel.com/oneapi/documentation/connect-with-ssh-linux-macos/

seems fpga is not available anymore..