vortexgpgpu/vortex #203: There is a Platform Config Makefile for Xilinx, but not for Altera

msaideroglu, 2024-11-14:
There is a Platform config Makefile for Xilinx xrt named **platforms.mk**, but not for Altera opae. When I try to run opae synthesis in DevCloud, it gives missing DPLATFORM_MEMORY_BANKS, DPLATFORM_MEMORY_ADDR_WIDTH error in synthesis report. Please check opae Makefile and give a proper platform configuration for Arria10 and Stratix10. Also DevCloud will be closed. Please provide a guide to synthesize the Vortex GPGPU on a Arria10 and Stratix10 DevKit with Intel OpenCL drivers instead of OPAE drivers. Thanks.

tinebp, 2024-11-14:
fixed in https://github.com/vortexgpgpu/vortex/commit/8230b37411dfe28fe1b59a25a5de4c7de276cf90

closed 2024-11-14 by tinebp

msaideroglu, 2024-11-14:
I cloned it and reinstalled. However 
`vortex/build/hw/syn/altera/opae$ PREFIX=test1 TARGET=fpga NUM_CORES=4 make`
still gives @tinebp 

> Critical Warning (13432): Verilog HDL Compiler Directive warning at local_mem_cfg_pkg.sv(49): text macro "PLATFORM_MEMORY_ADDR_WIDTH" is undefined File: /home/u229371/Desktop/GPGPU/Vortex_GPGPU/vortex_v2.2/build/hw/syn/altera/opae/test1_arria10_fpga_4c/src/local_mem_cfg_pkg.sv Line: 49
> Error (13411): Verilog HDL syntax error at local_mem_cfg_pkg.sv(49) near text / File: /home/u229371/Desktop/GPGPU/Vortex_GPGPU/vortex_v2.2/build/hw/syn/altera/opae/test1_arria10_fpga_4c/src/local_mem_cfg_pkg.sv Line: 49
> Error (13411): Verilog HDL syntax error at local_mem_cfg_pkg.sv(50) near text ; File: /home/u229371/Desktop/GPGPU/Vortex_GPGPU/vortex_v2.2/build/hw/syn/altera/opae/test1_arria10_fpga_4c/src/local_mem_cfg_pkg.sv Line: 50
> Critical Warning (13432): Verilog HDL Compiler Directive warning at local_mem_cfg_pkg.sv(52): text macro "PLATFORM_MEMORY_BURST_CNT_WIDTH" is undefined File: /home/u229371/Desktop/GPGPU/Vortex_GPGPU/vortex_v2.2/build/hw/syn/altera/opae/test1_arria10_fpga_4c/src/local_mem_cfg_pkg.sv Line: 52
> Error (13411): Verilog HDL syntax error at local_mem_cfg_pkg.sv(52) near text ; File: /home/u229371/Desktop/GPGPU/Vortex_GPGPU/vortex_v2.2/build/hw/syn/altera/opae/test1_arria10_fpga_4c/src/local_mem_cfg_pkg.sv Line: 52
> Warning (13228): Verilog HDL or VHDL warning at local_mem_cfg_pkg.sv(47): parameter declared inside package 'local_mem_cfg_pkg' shall be treated as localparam File: /home/u229371/Desktop/GPGPU/Vortex_GPGPU/vortex_v2.2/build/hw/syn/altera/opae/test1_arria10_fpga_4c/src/local_mem_cfg_pkg.sv Line: 47
> Error (13414): Verilog HDL error at local_mem_cfg_pkg.sv(49): system call "clog2" must have exactly one argument File: /home/u229371/Desktop/GPGPU/Vortex_GPGPU/vortex_v2.2/build/hw/syn/altera/opae/test1_arria10_fpga_4c/src/local_mem_cfg_pkg.sv Line: 49
> Warning (13228): Verilog HDL or VHDL warning at local_mem_cfg_pkg.sv(49): parameter declared inside package 'local_mem_cfg_pkg' shall be treated as localparam File: /home/u229371/Desktop/GPGPU/Vortex_GPGPU/vortex_v2.2/build/hw/syn/altera/opae/test1_arria10_fpga_4c/src/local_mem_cfg_pkg.sv Line: 49
> Error (13406): Verilog HDL error at local_mem_cfg_pkg.sv(55): object "LOCAL_MEM_DATA_WIDTH" is not declared File: /home/u229371/Desktop/GPGPU/Vortex_GPGPU/vortex_v2.2/build/hw/syn/altera/opae/test1_arria10_fpga_4c/src/local_mem_cfg_pkg.sv Line: 55
> Warning (13228): Verilog HDL or VHDL warning at local_mem_cfg_pkg.sv(55): parameter declared inside package 'local_mem_cfg_pkg' shall be treated as localparam File: /home/u229371/Desktop/GPGPU/Vortex_GPGPU/vortex_v2.2/build/hw/syn/altera/opae/test1_arria10_fpga_4c/src/local_mem_cfg_pkg.sv Line: 55
> Error (13406): Verilog HDL error at local_mem_cfg_pkg.sv(62): object "LOCAL_MEM_DATA_WIDTH" is not declared File: /home/u229371/Desktop/GPGPU/Vortex_GPGPU/vortex_v2.2/build/hw/syn/altera/opae/test1_arria10_fpga_4c/src/local_mem_cfg_pkg.sv Line: 62
> Error (17457): Verilog HDL error at local_mem_cfg_pkg.sv(62): range must be bounded by constant expressions File: /home/u229371/Desktop/GPGPU/Vortex_GPGPU/vortex_v2.2/build/hw/syn/altera/opae/test1_arria10_fpga_4c/src/local_mem_cfg_pkg.sv Line: 62
> Error (13406): Verilog HDL error at local_mem_cfg_pkg.sv(64): object "LOCAL_MEM_BURST_CNT_WIDTH" is not declared File: /home/u229371/Desktop/GPGPU/Vortex_GPGPU/vortex_v2.2/build/hw/syn/altera/opae/test1_arria10_fpga_4c/src/local_mem_cfg_pkg.sv Line: 64
> Error (17457): Verilog HDL error at local_mem_cfg_pkg.sv(64): range must be bounded by constant expressions File: /home/u229371/Desktop/GPGPU/Vortex_GPGPU/vortex_v2.2/build/hw/syn/altera/opae/test1_arria10_fpga_4c/src/local_mem_cfg_pkg.sv Line: 64
> Error (13363): Verilog HDL error at local_mem_cfg_pkg.sv(69): package "local_mem_cfg_pkg" ignored due to previous errors File: /home/u229371/Desktop/GPGPU/Vortex_GPGPU/vortex_v2.2/build/hw/syn/altera/opae/test1_arria10_fpga_4c/src/local_mem_cfg_pkg.sv Line: 69