vortexgpgpu/vortex #247: Build the toolchain from source.

ygttt, 2025-05-07:
Hello! Since I would like to make modifications to llvm-vortex, I need to build the toolchain from source. However, the documentation I’ve found so far is incomplete, and the project only provides a precompiled toolchain. Would it be possible for you to provide a document or guide for compiling the toolchain from source?

Udit8348, 2025-05-08:
You can install the pre-built toolchain. Then, clone in the same directory as your toolchain, but give it a more descript name such as llvm-vortex-build etc. You can go ahead and build it. The readme steps might be accurate, but I followed something like this:

```
git clone --recursive --branch vortex_2.x https://github.com/vortexgpgpu/llvm.git llvm-vortex-custom
cd llvm-vortex-custom
mkdir build && cd build
export LLVM_PREFIX=<path>/llvm-vortex-custom-build #this is where the build files will be written to
cmake -G "Unix Makefiles" -DLLVM_ABI_BREAKING_CHECKS=FORCE_OFF -DLLVM_INCLUDE_EXAMPLES=OFF -DLLVM_INCLUDE_TESTS=OFF -DCMAKE_BUILD_TYPE=Release -DLLVM_ENABLE_PROJECTS="clang" -DCMAKE_INSTALL_PREFIX=$LLVM_PREFIX -DDEFAULT_SYSROOT=$RISCV_TOOLCHAIN_PATH/riscv32-unknown-elf -DLLVM_DEFAULT_TARGET_TRIPLE="riscv32-unknown-elf" -DLLVM_TARGETS_TO_BUILD="RISCV" ../llvm
cmake -G "Unix Makefiles" -DLLVM_ABI_BREAKING_CHECKS=FORCE_OFF -DLLVM_INCLUDE_EXAMPLES=OFF -DLLVM_INCLUDE_TESTS=OFF -DBUILD_SHARED_LIBS=True -DCMAKE_BUILD_TYPE=RelWithDebInfo -DLLVM_ENABLE_ASSERTIONS=ON -DLLVM_ENABLE_PROJECTS="clang" -DCMAKE_INSTALL_PREFIX=$LLVM_PREFIX -DDEFAULT_SYSROOT=$RISCV_TOOLCHAIN_PATH/riscv32-unknown-elf -DLLVM_DEFAULT_TARGET_TRIPLE="riscv32-unknown-elf" -DLLVM_TARGETS_TO_BUILD="X86;RISCV;NVPTX" ../llvm
make -j`nproc`
make install
```

Then modify the path of `LLVM-VORTEX` to point to the path of the build folder. You see this variable in this file: https://github.com/vortexgpgpu/vortex/blob/master/config.mk.in

Then, you should rerun all the config steps and build vortex. You should see `which llvm-config` reports the path of your llvm-vortex that you built from source, and not the one from the prebuilt toolchain.

This is roughly the process that I can think of but there might be some additional debugging you need to do

tinebp, 2026-06-14:
`docs/building_toolchain.md` now documents building each toolchain component from source — Verilator, the RISC-V GNU toolchain, **LLVM for Vortex** (Clang + LLVM with the Vortex ISA extensions), compiler-rt, musl libc, POCL, and the optional tools. For modifying `llvm-vortex` specifically, build component 3 (LLVM for Vortex) and point your environment at that build. Closing.